pimoroni/enviroplus-python #3: base example gives error

koaning, 2019-06-24:
right after a clean install on rpi3 

```
(env) pi@raspberrypi:~/AIY-projects-python/enviroplus-python/examples $ python compensated-temperature.py 
compensated-temperature.py - Use the CPU temperature
to compensate temperature readings from the BME280 sensor.
Method adapted from Initial State's Enviro pHAT review:
https://medium.com/@InitialState/tutorial-review-enviro-phat-for-raspberry-pi-4cd6d8c63441

Press Ctrl+C to exit!

Traceback (most recent call last):
  File "compensated-temperature.py", line 39, in <module>
    cpu_temp = get_cpu_temperature()
  File "compensated-temperature.py", line 29, in get_cpu_temperature
    return float(output[output.index('=') + 1:output.rindex("'")])
TypeError: a bytes-like object is required, not 'str
```

koaning, 2019-06-24:
same story for python3 it seems 

```
(env) pi@raspberrypi:~/AIY-projects-python/enviroplus-python/examples $ python3 compensated-temperature.py 
compensated-temperature.py - Use the CPU temperature
to compensate temperature readings from the BME280 sensor.
Method adapted from Initial State's Enviro pHAT review:
https://medium.com/@InitialState/tutorial-review-enviro-phat-for-raspberry-pi-4cd6d8c63441

Press Ctrl+C to exit!

Traceback (most recent call last):
  File "compensated-temperature.py", line 39, in <module>
    cpu_temp = get_cpu_temperature()
  File "compensated-temperature.py", line 29, in get_cpu_temperature
    return float(output[output.index('=') + 1:output.rindex("'")])
TypeError: a bytes-like object is required, not 'str'
```

Gadgetoid, 2019-06-25:
Thanks for pointing this out. Hopefully fixed! I'm caught up in the Pi 4 chaos at the moment.

Out of interest, what do you get if you `python --version` since I would have expected this to work in Python 2.x.

kossmac, 2019-06-25:
with python3.7 it works if you put this to line 29:
`output = output.decode('utf-8')`

koaning, 2019-07-16:
i can confirm that it now works on my raspberry. thanks!

closed 2019-07-16 by koaning