pimoroni/enviroplus-python #6: Apparently incompatabile with Buster

hubertron, 2019-07-02:
Trying to hello world I get the follow errors

`Installed /usr/local/lib/python2.7/dist-packages/i2cdevice-0.0.5-py2.7.egg
Finished processing dependencies for enviroplus==0.0.1
Installing for Python 3..
Traceback (most recent call last):
  File "setup.py", line 26, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 28, in <module>
    from distutils.core import setup
ModuleNotFoundError: No module named 'distutils.core'
Backing up /boot/config.txt
Setting up i2c and SPI..
Setting up serial for PMS5003..
Done!`

Gadgetoid, 2019-07-03:
I'm guessing this is the Raspbian Lite flavour of Buster? Python, at least, should be installed by default  but - indeed - `numpy` and `smbus` should go into the installer.

hubertron, 2019-07-03:
Correct. Using the lite. It wasn't specified and knowing that intent for this is in a "headless" setup I didn't think I needed the full desktop.

Another concern is when trying to install Pillow via PIP I got a out of memory error and had to build a swap in order to install. I believe Pillow comes with the desktop Buster so maybe just update documentation that the full desktop version is required?

Gadgetoid, 2019-07-03:
Yeah- it absolutely should be good to go for a headless setup, i would be silly not to.

Pillow is one of the many Python packages that's paradoxically better served by apt, basically anything with significant C bindings is an uncompromising nightmare to install via pip. While there's a project to build Raspberry Pi-compatible wheels for all Python packages - https://www.piwheels.org/ - it only serves a subset of Python versions.

Looks like I need to include `python-pil` and `python3-pil`. You can install these with `sudo apt install python-pil python3-pil` albeit it's probably too late now.

Thanks for being an unwitting beginner and helping bash the rough edges off this installer- it's easy to lose sight of the wood for the trees when you've done this stuff for so long!

hotplot, 2019-07-06:
It appears that the supplied install scripts depend on `configparser` and the all-in-one example depends on `RPi.GPIO` and `spidev`, none of which are installed by default on buster lite. Can they also be added as dependencies and installed automatically?

To get the examples running I had to execute the following before being able to run the examples:

1) `sudo apt install python-pip`
2) `pip install configparser`
3) `pip3 install RPi.GPIO`
4) `pip3 install spidev`

(for added confusion, `configparser` is required by python2 while most people will be using python3 for everything else!)

Gadgetoid, 2019-07-06:
Thank you- I'll get these added ASAP. I'm hoping to roll the same installer out across other products, so this wisdom will be shared.

I'd swear Lite has become even lighter as of late, and it's certainly not a route I'd point beginners. Raspbian Desktop can be run headless by setting console startup mode in `raspi-config` and skirts some of the endless rigamarole of identifying and installing dependencies.

Alas there's not much I can do about the python2/python3 duality, although I do wonder if it's time to bite the tech support bullet and start forcing users to learn `python3` in lieu of `python`. On earlier releases it's not safe to assume that `python3` is even installed, and installing an entire additional runtime always felt a bit bruteforce to me.

WitheredHope, 2019-07-10:
This probably isn't a recommended work around but for me, after installing git on my buster lite install, I changed the line ```CONFIG_VARS=`python - <<EOF``` to python3 and everything worked for me with no need to install pillow, after that all I needed to manually install was spidev and RPi.GPIO and almost all of the examples ran fine. The only ones that didn't were all-in-one.py and luftdaten.py which for what I want to do isn't an issue for me.

tomgidden, 2019-07-22:
Also, at least on Buster (Pi Zero W, brand new install of Buster Lite), make sure user is in group `video` to access `vcgencmd`, and `dialout` for the serial.  The default `pi` user is, but it's not particularly straightforward for a beginner to figure out which device it needs, and the error isn't informative.

I'd suggest that the script(s) can be made to work for either Python easily, and the installer would use Python 3 if it's installed; Python 2 if not; and install and use Python 3 if _neither_ are there. (?)

For an outside Luftdaten box, a Zero with Buster Lite on a small card and no screen is not unreasonable... full desktop Buster on a Zero is _not_ fun, and all that activity would reduce the lifetime (well, mean-time-before-filesystem-corruption) of it.

I've rolled my own script to not run the ST7735 (pointless outside and fouls the light reading within an enclosure), reduce the CPU temperature compensation factor (I have the pHAT on a ribbon cable to get it away from the Pi), send the appropriate values to Luftdaten, and send _all_ the values to MQTT for local logging and display.

Next I have to come up with an enclosure that'll be weatherproof _and_ deliver Lux readings. It'd be good to include an extension ribbon cable with the Enviro+, or at least offer one as an accessory.

cipy, 2019-08-02:
I also managed to install/use Buster Lite, with the below workaround

```
  $ sudo apt install python-configparser python-setuptools spi-tools python-spidev
  $ sudo apt install python3-setuptools python3-spidev  

```

plus a $ git pull 

of the current code in the repo and a $ pip3 install RPi.GPIO

but Python3 compatibily should be investigated/confirmed, as it is not given #21

dxcSithLord, 2019-11-09:
Just finished running the install and I have been through [.](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-enviro-plus)  on a pi Zero with wifi and header.
`$ uname -a`
`Linux pi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux`
`$ lsb_release -a`
`No LSB modules are available.`
`Distributor ID:	Raspbian`
`Description:	Raspbian GNU/Linux 10 (buster)`
`Release:	10`
`Codename:	buster`

After running the install, which prompted to install the examples, which were in `~/Pimoroni/enviroplus/examples` and not `enviroplus-python/examles`, everything ran OK, until I ran the lcd.py.
`~/Pimoroni/enviroplus/examples $ python lcd.py`
`Traceback (most recent call last)`
`  File "lcd.py", line 3, in <module>`
`    import ST7735`
`  File "/usr/local/lib/python2.7/dist-packages/ST7735-0.0.3-py2.7.egg/ST7735/__init__.py", line 25, in <module>`
`    import spidev`
`ImportError: No module named spidev`
`~/Pimoroni/enviroplus/examples $ pip install spidev`
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple`
`Collecting spidev`
`  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz`
`Building wheels for collected packages: spidev`
`  Running setup.py bdist_wheel for spidev ... done`
`  Stored in directory: /home/pi/.cache/pip/wheels/10/d6/98/ba1f1999099e3e7adb3a0140af8c9287ff2b067ff005485674`
`Successfully built spidev`
`Installing collected packages: spidev`
`Successfully installed spidev-3.4`
`~/Pimoroni/enviroplus/examples $ python lcd.py`
`2019-11-09 18:11:45.914 INFO     lcd.py - Hello, World! example on the 0.96" LCD.`

`Press Ctrl+C to exit!`

I repeated this for python3:
`~/Pimoroni/enviroplus/examples $ python3 lcd.py`
`Traceback (most recent call last):`
`  File "lcd.py", line 3, in <module>`
`    import ST7735`
`  File "/usr/local/lib/python3.7/dist-packages/ST7735-0.0.3-py3.7.egg/ST7735/__init__.py", line 25, in <module>`
`    import spidev`
`ModuleNotFoundError: No module named 'spidev'`
`~/Pimoroni/enviroplus/examples $ pip3 install sdidev`
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple`
`Collecting sdidev`
`Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/sdidev/`

Thought you should know.

cipy, 2019-11-10:
@dxcSithLord have you tried my $ sudo steps listed just above your post here?

dxcSithLord, 2019-11-10:
Following your suggestion, I ran
`~/Pimoroni/enviroplus/examples $ sudo apt install python3-setuptools python3-spidev`
If I run lcd.py after that, I get a different error:
`python3 lcd.py`
`Traceback (most recent call last):`
`  File "lcd.py", line 3, in <module>`
`    import ST7735`
`  File "/usr/local/lib/python3.7/dist-packages/ST7735-0.0.3-py3.7.egg/ST7735/__init__.py", line 26, in <module>`
`    import RPi.GPIO as GPIO`
`ModuleNotFoundError: No module named 'RPi'`

`~/Pimoroni/enviroplus/examples $ pip3 install RPi.GPIO`
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple`
`Collecting RPi.GPIO`
`  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp37-cp37m-linux_armv6l.whl (69kB)`
`Installing collected packages: RPi.GPIO`
`Successfully installed RPi.GPIO-0.7.0`

`~/Pimoroni/enviroplus/examples $ python3 lcd.py`
`2019-11-10 22:10:22.021 INFO     lcd.py - Hello, World! example on the 0.96" LCD.`

`Press Ctrl+C to exit!`

So that worked.

jozefvaclavik, 2020-02-08:
I installed this on Buster Lite `2019-09-26` and `2020-02-05`.

I needed to install `sudo apt-get install git` so I can clone the repo and `pip install spidev` so I can run LCD examples. Everything else was installed by install script.

Darthmonkey, 2020-03-01:
I followed all the instructions including those above, lcd.py and all-in-one.py are working, but python luftdaten.py fails with:
```
Traceback (most recent call last):
  File "luftdaten.py", line 3, in <module>
    import requests
ImportError: No module named requests
```
Unless you run the following (I imagine there's a better option):
Traceback (most recent call last):
  File "luftdaten.py", line 3, in <module>
    import requests
ImportError: No module named requests
`pip install requests`

rockthy, 2020-07-11:
> I followed all the instructions including those above, lcd.py and all-in-one.py are working, but python luftdaten.py fails with:
> 
> ```
> Traceback (most recent call last):
>   File "luftdaten.py", line 3, in <module>
>     import requests
> ImportError: No module named requests
> ```
> 
> Unless you run the following (I imagine there's a better option):
> `pip install requests`

@Darthmonkey  I am getting this same issue, any fix for this yet?

closed 2023-01-20 by Gadgetoid