pimoroni/enviroplus-python #7: Issues with setup

sepulworld, 2019-07-04:
Working with Raspberry Lite

```
enviroplus 0.0.1 Python Library: Installer

Installing for Python 2..
Checking for python-numpy
Checking for python-smbus
Checking for python-pil
Traceback (most recent call last):
  File "setup.py", line 25, in <module>
    from setuptools import setup
ImportError: No module named setuptools
Installing for Python 3..
Checking for python3-numpy
Checking for python3-smbus
Checking for python3-pil
Traceback (most recent call last):
  File "setup.py", line 25, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'
Setting up i2c and SPI..
Backing up /boot/config.txt to /boot/config.preinstall-enviroplus-2019-04-04-07-04-05.txt
Setting up serial for PMS5003..
Adding dtoverlay=pi3-miniuart-bt # for Enviro+ to /boot/config.txt
Done!
```
Ill send in a PR to add python-setuptools

sepulworld, 2019-07-04:
Also noticed that I needed:

```
pip3 install pyserial
```

Gadgetoid, 2019-07-04:
Thank you. I've fixed the latter in https://github.com/pimoroni/pms5003-python/commit/fe59cec71125141ca975d4dfe8c855b59730f93d

I'm going to make `setuptools` a dependency in future release, and have included a version check to ensure it supports the metadata-in-setup.cfg technique that I use.

closed 2019-07-04 by Gadgetoid