pimoroni/enviroplus-python #9: pms5003 timeouts

sepulworld, 2019-07-04:
Hi, 

I keep getting timeouts with the pms5003 particle sensor library 

```
In [5]: readings = pms5003.read()
---------------------------------------------------------------------------
ReadTimeoutError                          Traceback (most recent call last)
<ipython-input-5-dc7f60780eda> in <module>
----> 1 readings = pms5003.read()

/usr/local/lib/python3.7/dist-packages/pms5003-0.0.2-py3.7.egg/pms5003/__init__.py in read(self)
    119             elapsed = time.time() - start
    120             if elapsed > 5:
--> 121                 raise ReadTimeoutError("PMS5003 Read Timeout")
    122
    123             sof = bytearray(self._serial.read(2))

ReadTimeoutError: PMS5003 Read Timeout
```

It times out every time. I have not gotten a reading from it yet. What are some things I can do to debug? I can hear the fan spinning on the pms5003 sensor, so it appears to be on.

torarne, 2019-07-04:
I have had the same problem - and in my case it was a loose connector in the cable - resetting it while pushing it in helped. Power is applied on a different pin to the comms.

Gadgetoid, 2019-07-04:
This will either be due to intermittent connectivity - as torarne mentions above - or could also be due to the `dtoverlay=pi3-miniuart-bt` missing from your `/boot/config.txt`. However from your other issue (seeing: "Adding dtoverlay=pi3-miniuart-bt # for Enviro+ to /boot/config.txt" in your install output) I believe it's more likely the former.

Gadgetoid, 2019-07-04:
Actually, looking back at my above post with a small amount of horror and the benefit of hindsight I do wonder if `config.txt` even supports inline comments at all. It could easily be that the `# for Enviro+` part is breaking that `dtoverlay` line altogether.

sepulworld, 2019-07-04:
Thanks for the quick feedback!

So, I first tried to reset the pms5003 cable connection and still had same issue.

I think removed the trailing comment `# for Enviro+` and now instead of a timeout I get...

```
particulates.py - Print readings from the PMS5003 particulate sensor.

Press Ctrl+C to exit!

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyAMA0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "particulates.py", line 12, in <module>
    pms5003 = PMS5003()
  File "/usr/local/lib/python3.7/dist-packages/pms5003-0.0.2-py3.7.egg/pms5003/__init__.py", line 93, in __init__
    self.setup()
  File "/usr/local/lib/python3.7/dist-packages/pms5003-0.0.2-py3.7.egg/pms5003/__init__.py", line 104, in setup
    self._serial = serial.Serial(self._device, baudrate=self._baudrate, timeout=4)
  File "/home/pi/.local/lib/python3.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/pi/.local/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyAMA0: [Errno 2] No such file or directory: '/dev/ttyAMA0'
```

ls of /dev
```
pi@enviro-pi1:~/enviroplus-python/examples $ ls /dev/
autofs           gpiomem  loop7               ptmx   ram6       stdout  tty2   tty32  tty45  tty58      urandom  vcsa2     vhci
block            hwrng    loop-control        pts    ram7       tty     tty20  tty33  tty46  tty59      v4l      vcsa3     video10
btrfs-control    i2c-1    mapper              ram0   ram8       tty0    tty21  tty34  tty47  tty6       vchiq    vcsa4     video11
bus              initctl  mem                 ram1   ram9       tty1    tty22  tty35  tty48  tty60      vcio     vcsa5     video12
cachefiles       input    memory_bandwidth    ram10  random     tty10   tty23  tty36  tty49  tty61      vc-mem   vcsa6     watchdog
char             kmsg     mmcblk0             ram11  raw        tty11   tty24  tty37  tty5   tty62      vcs      vcsm      watchdog0
console          log      mmcblk0p1           ram12  rfkill     tty12   tty25  tty38  tty50  tty63      vcs1     vcsm-cma  zero
cpu_dma_latency  loop0    mmcblk0p2           ram13  serial1    tty13   tty26  tty39  tty51  tty7       vcs2     vcsu
cuse             loop1    mqueue              ram14  shm        tty14   tty27  tty4   tty52  tty8       vcs3     vcsu1
disk             loop2    net                 ram15  snd        tty15   tty28  tty40  tty53  tty9       vcs4     vcsu2
fd               loop3    network_latency     ram2   spidev0.0  tty16   tty29  tty41  tty54  ttyprintk  vcs5     vcsu3
full             loop4    network_throughput  ram3   spidev0.1  tty17   tty3   tty42  tty55  ttyS0      vcs6     vcsu4
fuse             loop5    null                ram4   stderr     tty18   tty30  tty43  tty56  uhid       vcsa     vcsu5
gpiochip0        loop6    ppp                 ram5   stdin      tty19   tty31  tty44  tty57  uinput     vcsa1    vcsu6
```

Going to debug further. Perhaps the serial device that needs to be disabled isn't disabled properly?

sepulworld, 2019-07-04:
Here is my config.txt

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=dwc2

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=0
dtoverlay=pi3-miniuart-bt
```

Gadgetoid, 2019-07-04:
That `enable_uart=0` in your config should be an `enable_uart=1`, looks like your whole serial port is disabled rather than just the serial console.

sepulworld, 2019-07-05:
That did the trick! thank you!

closed 2019-07-05 by sepulworld