pimoroni/enviroplus-python #15: all-in-one error

koaning, 2019-07-16:
unfortunately there is a new error. when running this from ipython: 

```
In [12]: %run all-in-one.py                                                                                   
all-in-one.py - Displays readings from all of Enviro plus' sensors

Press Ctrl+C to exit!

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
~/AIY-projects-python/enviroplus-python/examples/all-in-one.py in <module>
    133             variable = "temperature"
    134             unit = "C"
--> 135             cpu_temp = get_cpu_temperature()
    136             # Smooth out with some averaging to decrease jitter
    137             cpu_temps = cpu_temps[1:] + [cpu_temp]

~/AIY-projects-python/enviroplus-python/examples/all-in-one.py in get_cpu_temperature()
     87     process = Popen(['vcgencmd', 'measure_temp'], stdout=PIPE)
     88     output, _error = process.communicate()
---> 89     return float(output[output.index('=') + 1:output.rindex("'")])
     90 
     91 

TypeError: a bytes-like object is required, not 'str'
```

koaning, 2019-07-16:
version information: 

```
In [15]: sys.version                                                                                          
Out[15]: '3.5.3 (default, Jan 19 2017, 14:11:04) \n[GCC 6.3.0 20170124]'
```

makeupsomething, 2019-08-01:
I was able to fix this by converting the output to a string

```python
decoded_output = output.decode("utf-8")
return float(decoded_output[decoded_output.index('=') + 1:decoded_output.rindex("'")])
```

cipy, 2019-08-02:
@koaning your iPython is version 3.5.3

and Python3 doesn't seem to be supported yet, see issue #21

kflmiami420, 2019-08-17:
Please can I get an estimated time this fix will be applied to the all_in_one_no_pm.py script so it can be run in python3  today is 08-17-2019

cipy, 2019-08-17:
@kflmiami420 well, the solution is to append `, universal_newlines=True` to all Popen calls

see #24

closed 2020-03-10 by Gadgetoid