pimoroni/enviroplus-python #16: trying to start it all on startup

koaning, 2019-07-16:
so ... this seems a bit odd. i have a script that runs just fine from the terminal but it gives this error when it is started up with `@reboot` in crontab. 

```
pms5003.ReadTimeoutError: PMS5003 Read Timeout
```

any suggestions what would be wise here? i am trying to make my raspberry log useful info to the screen such that when i leave it in a room with power plugged in it will show metrics without the need for the raspberry to be plugged into a screen.

koaning, 2019-07-16:
this is the contents of that python file: 

```
print("starting up")

import datetime as dt
import ST7735
from PIL import Image, ImageDraw, ImageFont
import time
from bme280 import BME280

time.sleep(10)
print("i have slept 10s")

try:
    from smbus2 import SMBus
except ImportError:
    from smbus import SMBus

# Create LCD class instance.
disp = ST7735.ST7735(
    port=0,
    cs=1,
    dc=9,
    backlight=12,
    rotation=270,
    spi_speed_hz=10000000
)

# Initialize display.
disp.begin()

# Width and height to calculate text position.
WIDTH = disp.width
HEIGHT = disp.height

# New canvas to draw on.
img = Image.new('RGB', (WIDTH, HEIGHT), color=(0, 0, 0))
draw = ImageDraw.Draw(img)

# Text settings.
font_size = 15
font = ImageFont.truetype("/home/pi/AIY-projects-python/enviroplus-python/examples/fonts/Asap/Asap-Bold.ttf", font_size)
text_colour = (255, 255, 255)
back_colour = (0, 170, 170)

import time
from pms5003 import PMS5003, ReadTimeoutError
bus = SMBus(1)
bme280 = BME280(i2c_dev=bus)
pms5003 = PMS5003()
t = 0
print("just before while loop:")

while True:
    t = (t + 1) % 2
    readings = pms5003.read()
    r1 = readings.pm_ug_per_m3(1)
    r2 = readings.pm_ug_per_m3(2.5)
    r10 = readings.pm_ug_per_m3(10)
    time.sleep(1.0)
    t = bme280.get_temperature()
    h = bme280.get_humidity()
            

    message = "1/2.5/10ug:{}/{}/{}\n".format(r1, r2, r10)
    message += "temp:{}C hum:{}%\n".format(round(t, 2), round(h, 1))
    message += "{}".format(str(dt.datetime.now())[:19])
    size_x, size_y = draw.textsize(message, font)

    # Calculate text position
    x = (WIDTH - size_x) / 2
    y = (HEIGHT / 2) - (size_y / 2)

    # Draw background rectangle and write text.
    draw.rectangle((0, 0, 160, 80), back_colour)
    draw.text((x, y), message, font=font, fill=text_colour)
    disp.display(img)
```

Gadgetoid, 2019-07-17:
Might be worth wrapping the call to `pms5003.read()` in a try/catch and handling the exception gracefully until it settles and starts working (assuming it does!).

```python
try:
    readings = pms5003.read()
    r1 = readings.pm_ug_per_m3(1)
    r2 = readings.pm_ug_per_m3(2.5)
    r10 = readings.pm_ug_per_m3(10)
except pms5003.ReadTimeoutError:
    r1 = 0
    r2 = 0
    r10 = 0
```

However, this is still bizarre. Are you sure you don't have any other services running at startup and using the GPIO?

cipy, 2019-08-02:
in my plays/tests data = pms5003.read() almost always gives a read timeout

as @Gadgetoid suggested, you @koaning should protect all read()'s with try's ;-)

for instance enviroplus-python/examples/luftdaten.py shows proper defensive programming

Gadgetoid, 2019-08-20:
See the updates `all-in-one.py` (thanks to @cipy) for an example of how to trap a read timeout on the `pms5003` without your entire script bailing out.

@cipy I took the liberty of replacing the `pass` with a `logging.warn`- we might as well make good use of logging while we have its awesome power!

cipy, 2019-08-24:
thanks @Gadgetoid for merging my PRs 👍

Gadgetoid, 2019-11-05:
I believe https://github.com/pimoroni/pms5003-python/issues/3 might be relevant to this issue and the fix - pending merge here: https://github.com/pimoroni/pms5003-python/pull/5 - probably applies also. By reading the SOF bytes one at a time, we should see fewer instances of read timeouts where the sensor isn't nicely synchronised should.

Closing for now, but please feel free to re-open if this is ongoing.

closed 2019-11-05 by Gadgetoid