pimoroni/enviroplus-python #17: First temperature reading always incorrect

roycepope, 2019-07-19:
It appears that the first value in each example is always incorrect. Why is that?

```console
root@zero:/home/pi/enviroplus-python/examples# ./temperature.py
21.6270364913
30.9992406745
30.9919257087
30.9839746596
30.9776138208
30.9737973178
30.9702988568
30.9661643122
```

```console
root@zero:/home/pi/enviroplus-python/examples# ./gas.py
gas.py - Print readings from the MICS6814 Gas sensor.

Oxidising: 72333.33 Ohms
Reducing: 21289.84 Ohms
NH3: 10811.28 Ohms

Oxidising: 617.65 Ohms
Reducing: 37759.51 Ohms
NH3: 461.96 Ohms

Oxidising: 565.66 Ohms
Reducing: 85284.40 Ohms
NH3: 721.92 Ohms

Oxidising: 513.76 Ohms
Reducing: 113696.97 Ohms
NH3: 1037.04 Ohms

Oxidising: 513.76 Ohms
Reducing: 135304.35 Ohms
NH3: 1409.13 Ohms

Oxidising: 565.66 Ohms
Reducing: 152108.11 Ohms
NH3: 1731.96 Ohms```

```console
root@zero:/home/pi/enviroplus-python/examples# ./light.py
light.py - Print readings from the LTR559 Light & Proximity sensor.

Light: 00.00 Lux
Proximity: 00.00

Light: 07.97 Lux
Proximity: 00.00

Light: 07.97 Lux
Proximity: 00.00

Light: 07.97 Lux
Proximity: 00.00```

Gadgetoid, 2019-07-22:
Most likely because in all cases the datasheet says something like "wait X amount of time for readings to stabilize" and I've not added a wait. Looks like I'll have to fix that.

Try putting a `time.sleep(1)` after the imports but before the main bulk of the code.

torarne, 2019-07-24:
I do not believe that this is purely due to the wait time, I do believe that the average calculation statements a re wrong - see the following I get by adding output of the new temp 👍 
Raw temperature: 20.95 *C
Compensated temperature: 36.03 *C
Raw temperature: 33.11 *C
Compensated temperature: 52.29 *C
Raw temperature: 33.11 *C
Compensated temperature: 40.92 *C
Raw temperature: 33.11 *C

If you look at the compensated temperature (from compensated-temperature.py) with one line added 
   print("Raw temperature: {:05.2f} *C".format(raw_temp))
    print("Compensated temperature: {:05.2f} *C".format(comp_temp))

You can see that the raw temperature is way too low at the start of the run (easily seen by restarting the code) and after a while the compensated temperature will converge and stabilise).

In other words - IMHO there are 2 bugs, one can be fixed by discarding the first raw temperature reading, the rest by fixing the compensation calculation.

cipy, 2019-08-02:
@roycepope for examples/compensated-temperature.py

you need to replace
cpu_temps = [0] * 5
with
cpu_temps = [get_cpu_temperature()] * 5

it is initiallized with (bogus) 0's, that's why you see wrong temperatures at very first 4 reads

cipy, 2019-08-02:
only these scripts seem to have bogus "buffers"

examples/all-in-one.py:cpu_temps = [0] * 5
examples/all-in-one-no-pm.py:cpu_temps = [0] * 5
examples/compensated-temperature.py:cpu_temps = [0] * 5

which means all those will give 4 wrong temperature measurements at start

regarding the other scripts for gas, light, etc; you should always discard the very first value you measure, if not even a few at the start of your "production" script, as these leading measurements can be influenced by physical constraints in the hardware/electronics: they/board need to reach a certain operating temperature, or voltage, or whatever, and you need to keep them in optimal operating (physical and electrical) parameters, otherwise you get random errors into your measurements.

So yeah, these Raspberry Pi (hardware) extensions are mostly for fun, if you need any (hard) precission, you need to grab a few math, physics and electronics books, study how these hardware extensions are made, from what materials and operating principles, and... someday your measurements will be more precise than today

 [get_cpu_temperature()] * 5 is a very good start @roycepope @torarne

cipy, 2019-08-04:
btw @Gadgetoid where can I/we see the datasheet? thanks,

Andermutu, 2019-08-13:
Hi, I have similar behaviour, but the temperature needs some readings to stabilize.  Here are my readings:

Compensated temperature: 32.65 *C
Compensated temperature: 55.50 *C
Compensated temperature: 41.95 *C
Compensated temperature: 28.56 *C
Compensated temperature: 15.03 *C
Compensated temperature: 15.34 *C
Compensated temperature: 15.63 *C
Compensated temperature: 15.76 *C
Compensated temperature: 15.73 *C
Compensated temperature: 15.83 *C
Compensated temperature: 15.80 *C
Compensated temperature: 15.92 *C
Compensated temperature: 16.04 *C
Compensated temperature: 16.31 *C
Compensated temperature: 16.43 *C
Compensated temperature: 16.55 *C
Compensated temperature: 16.81 *C
Compensated temperature: 16.78 *C
Compensated temperature: 16.91 *C
Compensated temperature: 16.90 *C
Compensated temperature: 17.01 *C
Compensated temperature: 17.39 *C
.........................................................
Compensated temperature: 19.46 *C #Correct temperature.

cipy, 2019-08-17:
@Andermutu see #25

closed 2019-08-19 by roycepope