pimoroni/enviroplus-python #18: celcius to farenheit?

Esteidinger, 2019-07-23:
Can anyone offer a suggestion on how to change the values to F? 

Thanks.

Gadgetoid, 2019-07-31:
`(1°C × 9/5) + 32 = 33.8°F`

If you want to get Farenheit directly out of the library then it might be worth raising a feature request against the bme280 driver here - https://github.com/pimoroni/bme280-python

kflmiami420, 2019-08-18:
Yes add the code and #comment it out and let users that want F  comment it  in.

Gadgetoid, 2019-08-20:
Good point!

While I think I'd prefer something like `bme280.get_temperature_fahrenheit()` or `bme280.get_temperature(unit='fahrenheit')` but since - in both instances - that would require the code being there and commented out... might as well stick the maths right into the examples.

kflmiami420, 2019-08-20:
This what I came up with :   

you have to add a mode or renumerate all your modes.   ( I found out that if you touch the light sensor it changes the display output ) 

I added a new value 

# Create a values dict to store the data
variables = ["temperature",
                   "temperature2",

Then I added a new mode called 1 since 0 is the default for Celsius degrees 

 # One mode for each variable
        if mode == 0:
            variable = "temperature"
            unit = "C"
            cpu_temp = get_cpu_temperature()
            # Smooth out with some averaging to decrease jitter
            cpu_temps = cpu_temps[1:] + [cpu_temp]
            avg_cpu_temp = sum(cpu_temps) / float(len(cpu_temps))
            raw_temp = bme280.get_temperature()
            data = raw_temp - ((avg_cpu_temp - raw_temp) / factor)
            display_text(variable, data, unit)

if mode == 1:
            variable = "temperature2"
            unit = "F"
            cpu_temp = get_cpu_temperature()
            # Smooth out with some averaging to decrease jitter
            cpu_temps = cpu_temps[1:] + [cpu_temp]
            avg_cpu_temp = sum(cpu_temps) / float(len(cpu_temps))
            raw_temp = bme280.get_temperature()
            data = raw_temp - ((avg_cpu_temp - raw_temp) / factor)
            data = (data * 1.8) + 32
            display_text(variable, data, unit)

I think my math is correct , Maybe someone can double check it .    

if it works for me then hopefully someone can incorporate the code in an update 

I am still unsure why my factor has top be adjusted so high to get proper temps ( I am at 8.2 factor)

 I have the pi zero w enclosed in a https://www.adafruit.com/product/3252 case with an extra tall  https://www.adafruit.com/product/1992 header so how do I remove the compensation code ? 

I want raw sensor temps and humidity ?

ScottTheHippy, 2022-04-09:
kflmiami420's code works almost- the last line of "display_text(variable, data, unit)" didn't work for me until I changed it to "display_text(variable[mode], data, unit)", then hey presto.  Many thanks!

closed 2023-01-20 by Gadgetoid