pimoroni/enviroplus-python #19: Converted c++ code into python

davidtme, 2019-07-29:
I'm trying to converted a bit of c++ code which uses a MiCS6814 to estimate the ppm of the detectable gasses but I'm not sure how to get the equivalent enviroplus-python values for resistance & base resistance with in the [getCurrentRatio](https://github.com/eNBeWe/MiCS6814-I2C-Library/blob/dc4efd9551938b7d2c132fa4e52a62e70618e61d/src/MiCS6814-I2C.cpp#L375) function which is then used with in the [measure](https://github.com/eNBeWe/MiCS6814-I2C-Library/blob/dc4efd9551938b7d2c132fa4e52a62e70618e61d/src/MiCS6814-I2C.cpp#L404) function.  Could anyone give me any pointers please?

Gadgetoid, 2019-07-31:
Looks like the code you're using talks - via i2c - to an ATmega168 which is what actually handles reading the gas sensor. I can't find the firmware for the ATmega168, so I can't figure out what it actually does.

If I were to hazard a guess, then the `baseResistance` is just the currently sensed system voltage since their schematic shows a `3V3_SENSE` line connected to an ADC channel. The "ratio" in this case, would then be the difference between the gas sensor output voltage and the system voltage.

IE if `3V3_SENSE` reads `2048` then we can assume that `2048==3.3v`, so if your gas reading reads `1024` and you divide that by `2048` you get `0.5` which you can assume means `1.65v`.

This is all pretty much guesswork, but I'd suggest that the `ratio` value can be calculated from our code by reading `adc.get_voltage('in0/gnd')` which gives you a voltage, and then dividing that by 3.3 to give you a ratio.

Hopefully working example:

```
from enviroplus import gas
gas.setup()
ox_ratio = gas.adc.get_voltage('in0/gnd') / 3.3
```

davidtme, 2019-07-31:
Thanks @Gadgetoid for spending some time on this, any help is much appreciated :)

Looking at [set_multiplexer](https://github.com/pimoroni/ads1015-python/blob/master/library/ads1015/__init__.py#L129) function, would I be able to use the following rather than `/ 3.3`

```
from enviroplus import gas
gas.setup()
ox_ratio = gas.adc.get_voltage('in0/ref')
```

Andermutu, 2019-08-13:
Hi @davidtme, @Gadgetoid;

I am also trying to measure gas concentrations, I need to get ppm values for CO, NO2, CH4... and the other detectable gases.

I think that the correct way to do it is as @Gadgetoid says, `ox_ratio = gas.adc.get_voltage('in0/gnd') / 3.3`. Doing by these way I can measure CO ppm (CO: 5.2351ppm)

But trying the same with `ox_ratio = gas.adc.get_voltage('in0/ref')` gives a negative ratio, so I would say it is wrong.

Thanks for your ideas and have a nice day.

davidtme, 2019-08-13:
@Andermutu I've tried the [adc.get_voltage('in0/gnd') / 3.3](https://github.com/davidtme/enviroplus-python/blob/master/examples/gas.py#L54) and I get some numbers :)

psycik, 2019-10-23:
This seems to work for me. Do wish I knew how accurate it was though.
`ox_ratio = gas.adc.get_voltage('in0/gnd') / 3.3
        red_ratio = gas.adc.get_voltage('in1/gnd') / 3.3
        nh3_ratio = gas.adc.get_voltage('in2/gnd') / 3.3
        co = pow(red_ratio, -1.179) * 4.385
        no2 = pow(ox_ratio, -1.007) * 6.855
        nh3 = pow(nh3_ratio, -1.67) * 1.47
        c3h8 = pow(nh3_ratio, -2.518) * 570.164
        c4h10 = pow(nh3_ratio, -2.138) * 398.107
        ch4 = pow(red_ratio, -4.363) * 630.957
        h2 = pow(red_ratio, -1.8) * 0.73
        c2h50h = pow(red_ratio, -1.552) * 1.622
        print( 'co: %s ppm' % co)
        print( 'no2: %s ppm' % no2)
        print( 'nh3: %s ppm' % nh3)
        print( 'c3h8: %s ppm' % c3h8)
        print( 'c4h10: %s ppm' % c4h10)
        print( 'ch4: %s ppm' % ch4)
        print( 'h2: %s ppm' % h2)
        print( 'c2h50h: %s ppm' % c2h50h)`

> co: 5.092087296056288 ppm
no2: 33.4387090215078 ppm
nh3: 2.0514111798875017 ppm
c3h8: 942.387572099747 ppm
c4h10: 609.9521573344416 ppm
ch4: 1097.1484880657024 ppm
h2: 0.9171637782593127 ppm
c2h50h: 1.9747757943785347 ppm

nophead, 2019-10-23:
I have two Enviro+ boards and the gas sensors give very different readings in the same room. Not only that they are extremely dependent on a few degrees change in room temperature. The OX sensor also has a slow climb over a month and seems no sign of stopping. I.e. started around 7K but is now 127K.

I don't see how any meaningful absolute readings can be made unless there is some temperature compensation and a calibration procedure to account for the large difference between sensors.

psycik, 2019-10-24:
> I have two Enviro+ boards and the gas sensors give very different readings in the same room. Not only that they are extremely dependent on a few degrees change in room temperature. The OX sensor [...]

Interesting, after 24 hours of running this code the values are still relatively close with no climb. I will keep it going but I'd imagine I would have seen a climb by now.

nophead, 2019-10-24:
This is what my first board has done over about 7 weeks, sampled every day. The cyan line is the OX sensor. My second board is on a similar trajectory.

[image]

nophead, 2019-10-24:
The code is here: https://github.com/nophead/EnviroPlusWeb