pimoroni/enviroplus-python #28: why use GPU instead of CPU for temperature compensation?

david-peterson, 2019-08-17:
Just out of curiosity - why does it compensate relative to the GPU temperature returned from the video core library instead of reading the CPU temperature? The two are pretty similar, but the latter is easier to get at.

https://github.com/pimoroni/enviroplus-python/blob/fcfd8a1d72993d2b0f09632c817f0234be0a7c47/examples/compensated-temperature.py#L27

You can directly read the CPU core temperature from /sys/class/thermal/thermal_zone0/temp and not run the vcgencmd sub process.

cipy, 2019-08-17:
maybe because the implementation was v0.9?

or maybe because vcgencmd has many other options/readings, that one can use at some point?

which readings are easier to fetch from vcgencmd than figururing out their /sys/class path? 🤪

Gadgetoid, 2019-08-19:
I imagine it was the command I was using during testing, so it just got baked into the script without any further consideration. Might well be worth changing.

closed 2020-03-16 by Gadgetoid