pimoroni/enviroplus-python #29: BME280 humidity readings issue

Andermutu, 2019-08-18:
Hi,

I am having issues with the humidity reading of the BME280 sensor. Here we have around 80% of relative humidity, however, the maximum value I get with the all-in-one-no-pm.py is 36-40% HR.

I have looked at the BME280 datasheets and it can read from 0-100% HR, so in theory, an 80% HR is achievable.

Any idea or suggestion is welcomed.

takkaria, 2019-08-18:
If you have mounted the hat directly onto the Pi, it's likely that the heat of the Pi is affecting your readings.  Try getting an extension cable and you should see better readings.

Andermutu, 2019-08-18:
Thank you @takkaria, I will try with an extension cable.

ADEidos, 2020-01-27:
Hi @Andermutu, did you finally tried with the extension cable?

doctorkb, 2021-10-06:
As mentioned, the heat is likely causing part of the issue.  I've opened another issue for updating the code to correctly calculate relative humidity after adjusting for CPU temps (e.g. as found in all_in_one and combined examples) -- you need to convert the RH to dewpoint using the raw temperature, then convert it back to RH using the compensated temperature.  I have some code that does this, but it's a little unpolished still. 🙂

sphaero, 2023-03-08:
Which issue is that? I'm also reading 18-36% humidity indoor with an extension cable.

sphaero, 2023-03-08:
I guess it was #104 and the code that's referred to is:

https://github.com/roscoe81/enviro-monitor/blob/ad8d3cec149dbc115825885bfebbf7fc1eb316cc/Northcliff_AQI_Monitor_Gen.py#L416