pimoroni/enviroplus-python #30: all-in-one.py always hangs after ltr559.get_proximity() returns 6291552

cipy, 2019-08-18:
and this 6291552 value is constant **every** single time

the python process hangs at random times too

> 2019-08-07 04:07:04.255 INFO     temp: 16.9 C
2019-08-08 06:27:57.758 INFO     temp: 17.6 C
2019-08-10 15:33:00.070 INFO     temp: 26.6 C
2019-08-11 22:16:45.719 INFO     temp: 21.5 C
2019-08-12 06:08:43.521 INFO     temp: 20.7 C
2019-08-13 03:26:02.298 INFO     temp: 19.0 C
2019-08-14 16:07:36.694 INFO     temp: 25.9 C
2019-08-15 04:02:06.605 INFO     temp: 17.3 C
2019-08-16 03:15:17.037 INFO     temp: 22.4 C
2019-08-17 17:53:19.751 INFO     temp: 26.6 C

anybody seen anything like it?

it must be an issue coming from ltr559, as the hole python process is hosed after that

cipy, 2019-08-18:
@takkaria do you have an advice / explanation for this? thanks,

takkaria, 2019-08-18:
I have not seen this myself

cipy, 2019-08-19:
interesting, when:

running python /usr/local/lib/python2.7/dist-packages/ltr559-0.0.4-py2.7.egg/ltr559/__init__.py
in parallel with python enviroplus-python/examples/all-in-one.py

> 2019-08-19 06:20:12.051 INFO     Lux: 346.93, Proximity: 0000
2019-08-19 06:20:12.212 INFO     Lux: 346.93, Proximity: 0013
2019-08-19 06:20:12.372 INFO     Lux: 346.93, Proximity: 0002
2019-08-19 06:20:12.531 INFO     Lux: 346.93, Proximity: 0000
2019-08-19 06:20:12.690 INFO     Lux: 346.93, Proximity: 0000
2019-08-19 06:20:12.852 INFO     Lux: 346.93, Proximity: 0000

>2019-08-19 06:20:12.125 INFO     temp: 22.3 C
('proxy now at', 6291552)
('mode now at', 24576)

only all*py hangs again. the ltr559 module still runs w/o apparent issues

cipy, 2019-08-28:
updating to the latest raspbian (buster) kernel doesn't seem to improve/change anything

cipy, 2019-09-20:
hmm, I used the proposed change from #31 to read the cpu temperature without spawning a process, but it continues to hang in the same place and with the same value 6291552

brettmlawson, 2021-12-14:
I am having this issue now. Did you ever find a resolution?