pimoroni/enviroplus-python #35: Put display to sleep (when using Luftdaten)

vschinazi, 2019-10-17:
Hi

I am really happy using my Enviro + and PMS5003 with Luftdaten but I was wondering if there is a way to put the display to sleep after a few minutes. The information  being displayed (ID and Wifi status) is useful for the initial configuration but not after that. 

I would love to learn more about this and how to edit the Luftdaten code to make this happen.

Thank you!

Gadgetoid, 2019-10-17:
Good idea! It should be relatively straightforward to accomplish too.

A simple approach would be to keep the time the script started and then- after a reasonable time frame- swap the call in the main loop from displaying status to another function that turns off the backlight and clears the display with a blank image.

A more complete solution might involve finding the SPI command to actually turn off the display and adding that upstream to the Python-st7735 library to eventually be called from the luftdaten example. Perhaps it could also turn back on if the proximity sensor is triggered?

The benefit of actually turning the display off versus just clearing to black would be a small saving in power.

vschinazi, 2019-10-21:
Thanks @Gadgetoid 
I really appreciate the suggestions. I am relatively new to the entire raspberry / coding world but I will see what I can do.

closed 2023-01-20 by Gadgetoid