pimoroni/enviroplus-python #40: rotate_mode

ShorTie8, 2019-11-01:
It be nice for me at least to see all the readings.
So I add a rotate_mode like proximity.
Anything > 0 enables it, default is 10 seconds for me.

`diff --git a/examples/all-in-one-no-pm.py b/examples/all-in-one-no-pm.py
index 6bb6873..100cf9e 100755
--- a/examples/all-in-one-no-pm.py
+++ b/examples/all-in-one-no-pm.py
@@ -101,6 +101,7 @@ factor = 0.8
 cpu_temps = [get_cpu_temperature()] * 5
 
 delay = 0.5  # Debounce the proximity tap
+rotate_mode = 10  # seconds between screen mode rotate
 mode = 0  # The starting mode
 last_page = 0
 light = 1
@@ -130,6 +131,12 @@ try:
             mode %= len(variables)
             last_page = time.time()
 
+        # If time elapsed > rotate_mode, show next mode
+        if rotate_mode > 0 and time.time() - last_page > rotate_mode:
+            mode += 1
+            mode %= len(variables)
+            last_page = time.time()
+
         # One mode for each variable
         if mode == 0:
             # variable = "temperature"
`

Gadgetoid, 2019-11-01:
This is a good idea- thank you!

ShorTie8, 2019-11-02:
Only problems it after awhile it does crash with

`Traceback (most recent call last):
  File "all-in-one-no-pm.py", line 126, in <module>
    proximity = ltr559.get_proximity()
  File "/usr/local/lib/python2.7/dist-packages/ltr559-0.1.0-py2.7.egg/ltr559/__init__.py", line 454, in get_proximity
    self.update_sensor()
  File "/usr/local/lib/python2.7/dist-packages/ltr559-0.1.0-py2.7.egg/ltr559/__init__.py", line 392, in update_sensor
    self._ps0 = self._ltr559.get('PS_DATA').ch0
IndexError: tuple index out of range
`
and I know nothing about Python .. :/~

Gadgetoid, 2019-11-05:
That's odd- I can't see anything in `update_sensor` ( in the `ltr559` library ) that should ever throw a tuple index error.

What do you see if you run:

```
python -c "import ltr559;print(ltr559.__version__)"
```

Or, alternatively:

```
pip show ltr559
```

ShorTie8, 2019-11-07:
pi@raspberrypi:~ $ python -c "import ltr559;print(ltr559.__version__)"
0.1.0
pi@raspberrypi:~ $ pip show ltr559
Name: ltr559
Version: 0.1.0
Summary: Python library for the LTR559 light and proximity
Home-page: http://www.pimoroni.com
Author: Philip Howard
Author-email: phil@pimoroni.com
License: MIT
Location: /usr/local/lib/python2.7/dist-packages/ltr559-0.1.0-py2.7.egg
Requires: i2cdevice
Required-by: enviroplus
pi@raspberrypi:~ $

Gadgetoid, 2019-11-07:
In retrospect that should have been obvious by the fact `ltr559-0.1.0` was right there in your original comment! Sometimes I can't see the wood for the trees.

I can't see any reason- at a glance- why that error would crop up. I'll run some tests.

closed 2023-01-20 by Gadgetoid