pimoroni/enviroplus-python #41: Combined.py - not python 3 compatible and text display issues

markmizu, 2019-11-14:
In the display_everything() function - 
xrange is deprecated in python 3
also there is a creep in the x position of each successive text element so they display in a stepped line going off the right side of the screen. The y positions are fine, just the x positions are wrong.

I fixed as follows - 
A: changed both xrange() statements to range()
B: changed end of x= statement from (i / row_count))  to  (i % column_count))

`# Displays all the text on the 0.96" LCD
def display_everything():
    draw.rectangle((0, 0, WIDTH, HEIGHT), (0, 0, 0))
    column_count = 2
    row_count = (len(variables)/column_count)
    for i in range(len(variables)):
        variable = variables[i]
        data_value = values[variable][-1]
        unit = units[i]
        x = x_offset + ((WIDTH/column_count) * (i % column_count))
        y = y_offset + ((HEIGHT/row_count) * (i % row_count))
        message = "{}: {:.1f} {}".format(variable[:4], data_value, unit)
        lim = limits[i]
        rgb = palette[0]
        for j in range(len(lim)):
            if data_value > lim[j]:
                rgb = palette[j+1]
        draw.text((x, y), message, font=smallfont, fill=rgb)
    st7735.display(img)`

nophead, 2019-11-14:
To fix the positions change all the / to // to get integer division on Python3.

markmizu, 2019-11-15:
@nophead Thanks. Yes replacing the / for // corrected that issue for me.
Switched end of that line to ...* ( i // row_count))

Gadgetoid, 2020-03-17:
Any ideas on this one @sandyjmacdonald - looks like you did some porting to Python 3 with fixes here https://github.com/pimoroni/enviroplus-python/commit/20442c9a53edb05b925faa07eb8360bd46442978 but the code mentioned above hasn't changed insofar as I can tell.

Gadgetoid, 2021-11-02:
Tested locally. Seems to be working.

closed 2021-11-02 by Gadgetoid