pimoroni/enviroplus-python #42: Luftdaten.py script stops sync with luftdaten.info after 3 days

espressobeanies, 2019-11-17:
I'm not sure how to troubleshoot this one further but I'm noticing the luftdaten.py script stops syncing with the https://luftdaten.info site after running continuously for 3 or more days straight. Only a restart of the script appears to resolve the issue. Please advise on how to diagnose further.

woll, 2019-11-26:
I've had a similar thing (using the Enviro+ I got via the Pint of Science event, which I assume is running software very similar to that in this github). It has stopped updating the luftdaten data twice in the few days since I installed it. 

The first time, I just turned the rapsberry pi off and on and it started working again. The second time, it stopped at around 4am. When I noticed, around 10am, it was still connected to the WiFi network and I could login to the raspberry pi via ssh. Not sure if was a problem with the WiFi at some time that may have caused the problem.

The script is started using this crontab entry:
   @reboot sleep 30; sudo python3 /home/pi/Pimoroni/enviroplus/examples/luftdaten-uos.py
but when I logged on, the python3 process was not running. So, for some reason the python3 process must have stopped/crashed. It really needs to be run using something like launchd to restart the script if it crashes.

woll, 2019-11-29:
I edited the crontab to save the log to a file. After crashing, this was the last line in the log file:
{'temperature': '1.64', 'pressure': '100732.78', 'humidity': '49.77', 'gas_oxidising': '953836.07', 'gas_reducing': '338871.79', 'gas_nh3': '811605.63', 'P2': '1', 'P1': '1'}free(): invalid pointer

woll, 2019-12-01:
In the luftdaten-uos.py script (from the software supplied with the Eniviro+ at the Pint of Science event) the stack trace when it crashes is:
{'temperature': '-2.11', 'pressure': '101490.39', 'humidity': '50.19', 'gas_oxidising': '87255.81', 'gas_reducing': '219000.00', 'gas_nh3': '417846.15', 'P2': '61', 'P1': '74'}free(): invalid pointer
Fatal Python error: Aborted

Current thread 0xb6efb8e0 (most recent call first):
  File "/usr/local/lib/python3.7/dist-packages/ST7735-0.0.3-py3.7.egg/ST7735/__init__.py", line 212 in send
  File "/usr/local/lib/python3.7/dist-packages/ST7735-0.0.3-py3.7.egg/ST7735/__init__.py", line 233 in data
  File "/usr/local/lib/python3.7/dist-packages/ST7735-0.0.3-py3.7.egg/ST7735/__init__.py", line 413 in display
  File "/home/pi/Pimoroni/enviroplus/examples/luftdaten-uos.py", line 130 in display_status
  File "/home/pi/Pimoroni/enviroplus/examples/luftdaten-uos.py", line 236 in <module>

woll, 2019-12-03:
Line 130 of luftdaten-uos.py is:
    disp.display(img)
inside the display_status() function.

Gadgetoid, 2019-12-03:
This is potentially more shenanigans with the Python imaging library (it's plaguing us on another project, too)- it will surely be tricky to debug since things start to get sketchy when the errors aren't really Python errors.

It might be worth commenting out the call to `display_status()` and seeing if it remains running for more than 3 days without display output:

https://github.com/pimoroni/enviroplus-python/blob/655d013499ca1b98a9d28884c7a8c2607972f09e/examples/luftdaten.py#L183

woll, 2019-12-10:
Yes, I commented out the display_status() call, and it has been running OK for over 1 week.

With something like this, that needs to be running forever, the script really needs to be started using launchd (or equivalent) with a KEEPALIVE setting , so it gets restarted if it crashes because of library problems.

espressobeanies, 2020-01-06:
I can confirm commenting out the display_status() function call resolves the issue. Thanks all!

closed 2020-01-06 by espressobeanies

gidhap, 2020-01-13:
I can confirm this solved the issue for me too :)