pimoroni/enviroplus-python #43: Please clarify the PMS output

ferazambuja, 2019-11-17:
The current output erroneously gives the impression that the first 3 values are the actual amount of particles read by the sensor with the names like combustion particles, organic compounds, metals.
Those values are actually the standard value, which is a calculation off the concentration "corrected" to the "standard atmosphere", and not the real values of the sensor.
https://www.aqmd.gov/docs/default-source/aq-spec/resources-page/plantower-pms5003-manual_v2-3.pdf

> PM1.0 ug/m3 (ultrafine particles):                             88
PM2.5 ug/m3 (combustion particles, organic compounds, metals): 129
PM10 ug/m3  (dust, pollen, mould spores):                      148
PM1.0 ug/m3 (atmos env):                                       58
PM2.5 ug/m3 (atmos env):                                       85
PM10 ug/m3 (atmos env):                                        98

There is more info here:
https://publiclab.org/questions/samr/04-07-2019/how-to-interpret-pms5003-sensor-values#c23772

At least for me, this is how different this values are comparing to other similar devices side by side reporting in "under atmospheric environment" :
https://imgur.com/a/NbRluZU

ADEidos, 2020-01-31:
Hi @ferazambuja ,
I am interesting in what you are poiting out about this.
I am using the PMS5003 sensor and I have some dubts about values I am getting with it.
They are too hight to be considered true.
I am also trying to find a way to verify the values but I didn't get a good answer so far.

nophead, 2020-01-31:
I don't think the PMS5003 measures temperature and pressure so I think all it can do is make assumptions for fan flow rate, air density and particle density to produce those "corrected" readings.

closed 2023-01-20 by Gadgetoid