pimoroni/enviroplus-python #46: Adding a CSV writer to the All-in-One script

Ci4D, 2020-01-08:
Could anyone help me with this? I tried to add a csv writer to the all-in-one script so that each row indicated the variables taken at each interval. However I couldnt get this to work properly. Does anyone have anything similar they could share? Thanks

hulleyrob, 2020-01-08:
example? using the logging module would do the job then just output comma separated variables if that is the format you are after. Not sure this is really an issue though.

Ci4D, 2020-01-08:
Sorry this might be the wrong place to ask. I am trying to write the values read for each measurement to a csv.

hulleyrob, 2020-01-08:
import csv

output = [[1,2,3,4],[5,6,7,8]]

with open('//Users/homefolder/csv_file.csv', mode='w') as csv_file:
    csv_output = csv.writer(csv_file, delimiter=',', quotechar='"', quoting=csv.QUOTE_MINIMAL)
    for a,b,c,d in output:
        csv_output.writerow([a,b,c,d])

something like that should be enough for you to get the idea

Ci4D, 2020-01-08:
Awesome, thanks a lot!