pimoroni/enviroplus-python #47: Understanding the MICS output

CalmArms, 2020-01-14:
Hey there, I really appreciate the work that's gone into the Enviro+ and the library, it's a nifty piece of kit!

I'm having a bit of trouble at the moment translating the readings from the MICS sensor into something meaningful beyond increasing or decreasing air quality relative to the environment it started measuring in.

Are there are any reference materials for translating the Ohms output by each of the sensors to some bands for measuring air quality? For example, if the Oxidising output was 4570 Ohms, what does that mean in itself? Is that normal levels? Dangerous levels? Or does it not intrinsically mean anything?

If the latter, am I right in thinking that to measure the air quality of a particular area, I first need to start the sensor in a "clean" room to set the baseline?

Any guidance is much appreciated!

nophead, 2020-01-14:
My experience is that the gas sensors don't give any meaningful absolute values because temperature, pressure and humidity massively affect the readings. 

[image]

Here you can see the gas sensors seem to be pretty much the inverse of humidity. Was there a massive increase in pollution, or simply a drop in humidity?

CalmArms, 2020-01-16:
Ah... well observed @nophead. I'll leave the issue open for a bit in case anybody else is able to pitch in, but I think you're probably right.

Gadgetoid, 2020-01-16:
This is- indeed- correct. These sensors don't produce much meaningful information without context- that may be either known baseline or calibration. You can use other sources to quantity a baseline measurement- IE: measure for a day and go and check other data for confirmation on what your measurements should be showing- and then use variation upon that measurement to infer data points from your future readings.

Regarding @nophead's comment about the gas displaying the inverse of humidity- I don't know off the top of my head, but the effects of other environmental factors on gas readings should be well documented and/or studied somewhere. I usually rely on @sandyjmacdonald and his super science background to dig these up and turn them into something a mere layperson such as myself can understand. 😆 

I think we can probably do better with explaining the shortfalls of these sensor packages and how best to use them, given they are effectively uncalibrated, raw sensors. In an ideal world they would be calibrated, but - at least in the case of gas sensors - this requires obtaining known gas mixes and exposing the installed sensor to them.

nophead, 2020-01-16:
This is a long term graph:

[image]

If you look at the oxi trace it goes from 10K, through 20K, which is supposed to be the nominal value, rises steadily over time and is now regularly over 400K. Note it got moved from the UK inside to Tenerife outside where the temperature jumps in the middle. I can't make any sense of the reading unless oxidising gases are steadily increasing over time.

closed 2023-01-20 by Gadgetoid