pimoroni/enviroplus-python #49: Data reading with PMS5003 - super hi values

ADEidos, 2020-01-27:
Hi,
I have some dubts about values I am getting with PMS5003 sensor.
I am getting very hight values (almoust the double) from the expected.
Is there a way to verify if the sensor is working properly?
The data I am getting are to hi to be very credible.
Is there in general a way to verify data from PMS5003 and BME280?
Thank you!

gidhap, 2020-01-27:
Same issue here, keen to see what happens

nophead, 2020-01-27:
I get super high values indoors from stir fry cooking and making the bed. Near Manchester I get moderate values from traffic on the road outside. Outside in a shed in Tenerife I only get low values.

You can see where I moved it from the increase in temperatures just left of centre.

[image]

gidhap, 2020-01-27:
well doesn't this look right? Stir fry cooking creates steam, making bed creates dust. Both are deflecting the laser in the PM sensor, causing high redings...

nophead, 2020-01-27:
Yes my values look right to me. I am showing how high it can be legitimately. What makes you think it is reading higher than it should be?

gidhap, 2020-01-27:
I am struggling with the PM values. As soon as there is fog, same effect like your steam, the sensor doesn't show the real particles as it is cheated by the fog. Professional solutions can compensate this effect. Ans I am comparing my values with the ones of two "official" PM stations nearby. OK and matching as long as there isn't too much humidity in the air.

nophead, 2020-01-27:
I don't think the effect of stir fry is steam because the sensor was in an upstairs bedroom not directly above the kitchen. All that would be noticeable in the bedroom would be a alight smell of cooking. Presumably that smell is micro particles of food.

I can see fog would be counted as it is micro droplets of water. Not sure how that is compensated for. Perhaps they heat the incoming air to evaporate any water. I know some sensors use a heater to cause a convective flow rather than using a fan.

Tojaj, 2020-02-09:
My sensor got crazy today, these are values I see indoor:

    2020-02-09 16:23:37.007 INFO     
    PM1.0 ug/m3 (ultrafine particles):                             509
    PM2.5 ug/m3 (combustion particles, organic compounds, metals): 851
    PM10 ug/m3  (dust, pollen, mould spores):                      871
    PM1.0 ug/m3 (atmos env):                                       339
    PM2.5 ug/m3 (atmos env):                                       566
    PM10 ug/m3 (atmos env):                                        579
    >0.3um in 0.1L air:                                            65535
    >0.5um in 0.1L air:                                            23484
    >1.0um in 0.1L air:                                            6945
    >2.5um in 0.1L air:                                            308
    >5.0um in 0.1L air:                                            19
    >10um in 0.1L air:                                             2

Last week, the values it reported were reasonable, then I had it turned off for about a week and when I tried it again, this is what I got. I wonder if any of you have the same experience?

jozefvaclavik, 2020-04-13:
I have fairly opposite experience. Recently purchased air purifier for indoors that comes with PM sensor of its own. Placed it in one room. Got values ~10ug/m3 once stabilized (and purified, lol). Placed it in another (small) room and the numbers went crazy high. I’m talking 230ug/m3. So I decided to compare this with PMS5003. In first room I got values similar to air purifier. But in the other room just slightly higher. 15ug/m3. Thats seems pretty big difference against 230ug/m3.

To be fair, the placement wasn’t same, but I wouldn’t expect it that to have such major difference.

I’ll keep digging/comparing.

@Tojaj 65535 sounds like max int value.

closed 2023-01-20 by Gadgetoid