pimoroni/enviroplus-python #55: Localization of dt in sun_moon_time() of weather-and-light.py

roscoe81, 2020-03-12:
I've found that the dt = pytz.timezone(time_zone).localize(dt) in line 90 appears to always return the datetime based on the Raspberry Pi's timezone setting , rather than the timezone provided to pytz. This issue is highlighted here https://www.journaldev.com/23370/python-pytz

This is only an issue if the timezone provided to pytz differs from the Raspberry Pi's timezone setting. In may case, I want to be able to deploy systems in multiple time zones without needing to change the Raspberry Pi's timezone setting.

I addressed it by using astimezone to convert to the desired timezone via UTC as follows:
    utc = pytz.utc
    utc_dt = datetime.now(tz=utc)
    local_dt = utc_dt.astimezone(pytz.timezone(time_zone))
    today = local_dt.date()
I then used local_dt in the remainder of sun_moon_time() and also returned that value so that the local time is displayed, rather than the Raspberry Pi's system time.

Let me know if you agree that this is an issue. If so, I'm happy to raise a PR.

Gadgetoid, 2020-03-12:
A PR would be great, thank you!- this seems like a reasonable conclusion to me. I believe this code was written by @sandyjmacdonald so I'll poke him to review the PR also.

roscoe81, 2020-03-12:
Done!

Gadgetoid, 2020-03-13:
Thank you! I'll slide it under Sandy's nose for a approval and merge ASAP!

closed 2020-03-13 by Gadgetoid