pimoroni/enviroplus-python #58: Extra dependency needed

C-D-Lewis, 2020-03-24:
With the regular white Enviro (not with air monitoring), I needed to install one more dependency to get the noise-profile.py working. Possibly this should be done when enviro is setup the first time.

```
python3 -m pip install sounddevice
```

Gadgetoid, 2020-03-30:
Thank you. This is now in `install_requires` and- I think- released as of `0.0.3` - https://github.com/pimoroni/enviroplus-python/blob/master/library/setup.cfg#L40

closed 2020-03-30 by Gadgetoid

MatthewWilkes, 2020-05-23:
@Gadgetoid Hi there! Sorry to re-open this, but I think it's relevant. The 0.0.3 release does *not* include sounddevice, as it has a `install_requires=` line in setup.py which is overriding the values from `setup.cfg`.

This was fixed in `4cc6c622eb92bbc223db977e37bd351f84ef2ee5` but that was 8 days after version 0.0.3 was released. Would you cut a new release please?