pimoroni/enviroplus-python #59: RobotMedium fonts.ttf not found

dxcSithLord, 2020-03-30:
from fonts.ttf import RobotMedium as UserFont

is in combined.py, but fonts.ttf is not installed by this package.

Gadgetoid, 2020-03-30:
You'll need to install `fonts` and `font-roboto` via pip (or pip3).

These are specified in `install_requires` but it's possible that hasn't made it to a release yet:

https://github.com/pimoroni/enviroplus-python/blob/97ee1d88e84df79d82c3ca20eb89378c65197f9a/library/setup.cfg#L36-L37

dxcSithLord, 2020-03-30:
I was updating manually from an earlier version - I should have used RobotoMedium, but the fonts and font-roboto were both missing.  This can be closed, thank you.

Gadgetoid, 2020-03-31:
No worries!

closed 2020-03-31 by Gadgetoid

shtefanilie, 2020-04-04:
I still have that issue. Restarted, uninstalled countless times... Am I doing smth wrong?

```
pip3 install font-roboto
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: font-roboto in /usr/local/lib/python3.7/dist-packages/font_roboto-0.0.1-py3.7.egg (0.0.1)
```

```
Traceback (most recent call last):
  File "weather-and-light.py", line 9, in <module>
    from fonts.ttf import RobotoMedium as UserFont
ImportError: cannot import name 'RobotoMedium' from 'fonts.ttf' (/usr/local/lib/python3.7/dist-packages/fonts-0.0.3-py3.7.egg/fonts/ttf/__init__.py)
```

dxcSithLord, 2020-04-04:
Have you done
`pip3 install fonts`
?

shtefanilie, 2020-04-05:
@dxcSithLord yes 🤔 
```
pip3 install fonts

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: fonts in /usr/local/lib/python3.7/dist-packages/fonts-0.0.3-py3.7.egg (0.0.3)
```

shtefanilie, 2020-04-05:
Nvm, i've purged everything, remade a fresh `raspbian` SD, installed via curl `curl -sSL https://get.pimoroni.com/enviroplus | bash`, and everything's fine. More than likely it was some old assets.