pimoroni/enviroplus-python #63: Install misses package

java-artisan, 2020-05-02:
The install fails because it can't find a package named 'ffi'. For both Python 2 and 3.

This is fixed by running:
`sudo apt-get install python3-ccfi python-cffi`
And restarting the install.

Hellspawny, 2020-05-03:
Sadly, python3-ccfi isn't available at the moment.

`sudo apt-get install python3-ccfi`
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python3-ccfi

java-artisan, 2020-05-03:
My mistake, that's a typo. I copy/pasted the Python 2 package name for Python 3. The instruction should read:
`sudo apt-get install python3-cffi python-cffi`

Hellspawny, 2020-05-03:
Thank you, should have recognized it myself.

closed 2020-07-30 by Gadgetoid