pimoroni/enviroplus-python #65: problems with enviro + library

belidi, 2020-05-14:
I have a problem with the enviro (+) library with an enviro card. When I launch weather.py I get these messages:
  Traceback (most recent call last):
   File "/home/pi/enviroplus-python/examples/weather.py", line 4, in <module>
     from bme280 import BME280
   File "/usr/local/lib/python3.7/dist-packages/pimoroni_bme280-0.0.2-py3.7.egg/bme280/__init__.py", line 2, in <module>
     from i2cdevice import Device, Register, BitField, _int_to_bytes
ModuleNotFoundError: No module named 'i2cdevice'

I did ; pip install i2cdevice and bme280, but always the same,

StarlightLab, 2020-05-15:
Same issue.  Hoping to find out if something else uses SMBus at the same time??

Gadgetoid, 2020-05-15:
This is odd, since the `bme280` module *requires* i2cdevice, and installation should fail where it's not available.

In your case you're running under Python 3, so you should try:

```
sudo pip3 install i2cdevice
```

Do you have any further information about the OS version you're running, any errors encountered in the install process or other clues that could reveal how your system got into this situation?

belidi, 2020-05-16:
The correction of i2cdevice is done, thanks for the correction (pip3). There remains now the problem of the BME280 module which causes the following error:
Traceback (most recent call last):
   File "/home/pi/enviroplus-python/examples/weather.py", line 4, in <module>
     from bme280 import BME280
ImportError: cannot import name 'BME280' from 'bme280' (/usr/local/lib/python3.7/dist-packages/bme280/__init__.py)

pi zero W
Raspbian Buster with desktop 4.19.97+ # 1294
installation of the library by github (git clone ...).

nophead, 2020-05-16:
I think you need the Pimoroni version of the BME library where BME280 is a class. On my system it is pimoroni_bme280-0.0.2-py3.7.egg

belidi, 2020-05-18:
after doing: sudo pip3 install pimoroni-bme280, it remains the same always the same error message.should i start from a new installation: os + library

nophead, 2020-05-18:
Perhaps you just need to uninstall the other version you had before. I don't know what search order it uses.

aussieEd, 2020-10-31:
I am also having the same issue and have tried all of the above suggestions. Is there anything further I can try to get this working?

Other sensors on the Enviro+ appear to be working and it is just this one which is not.

daniel-rhoades, 2020-11-01:
> I am also having the same issue and have tried all of the above suggestions. Is there anything further I can try to get this working?
> 
> Other sensors on the Enviro+ appear to be working and it is just this one which is not.

@aussieEd - There appears to be an issue with v0.1.0 of the Pimoroni BME280 library, see https://github.com/pimoroni/bme280-python/issues/12 along with a workaround

closed 2023-01-20 by Gadgetoid