pimoroni/enviroplus-python #66: Easy way to detect if gas sensor is available

David-Hari, 2020-05-15:
Hello,

I have both an Enviro (for inside) and an Enviro+ (for outside).

I am writing code to collect data from them. The same code will run on each device, with the only difference being that the Enviro+ also collects data from the gas sensor.

At the moment, I am using this rather nasty bit of code to detect if the gas sensor is available:
```
try:
    gas.read_all()
    hasGasSensor = True
except Exception:
    hasGasSensor = False
```

This works, however I am worried about what kind of exceptions might be thrown during ```read_all```. Without the try/except, I get "OSError: [Errno 121] Remote I/O error" from the i2c code. I only hope that code is properly cleaning up after it fails to connect to the sensor, or that it doesn't modify anything.

It would be nice if the library provided a cleaner way to detect if the sensor is available.

angelafevi95, 2020-05-24:
Hi Davi-Hari, 

Were you able to solve the error mentioned without applying that exception? 
I am unable to get any reading of gas.py example due to the same issue.

KR

David-Hari, 2020-05-25:
No, this is a feature request (of sorts).

I can do the check just fine using exception handling. I just find that a bit icky. It would be better if the library could provide a way to check if the sensor is attached without having to try and read from it.

Gadgetoid, 2020-06-18:
Insofar as I'm aware there's no way to check for the existence of an i2c device other than to attempt to read/write it- so the code you're using is correct. I agree that it would be worth having as a library feature, though.

PR #74 (2021-11-02): Add available() method to gas sensor for #66
This change catches an IOError when setting up the gas sensor and provides an `available()` method for determining if a sensor is present. Aims to fix #66
closed 2021-11-02 by Gadgetoid