pimoroni/enviroplus-python #76: Suggested enhancement - map gas readings to values stated in MiCS-6814 data sheet

MetPhoto, 2020-07-01:
Total newbie suggestion. I am no Pythong programmer. 

Why not map the raw values in KOhms from the gas sensor to the value range shown in the data sheet for the MiCS-6814?

For example:
```
from numpy import interp

# Nitrogen dioxide
oxidised_data = interp(oxidised_data,[0.8,20],[0.05,10])

# Carbon monoxide
reduced_data = interp(reduced_data,[100,1500],[1,1000])

# Amonia.
nh3_data = interp(nh3_data,[10,1500],[1,300])
```

This code does not take into sensitivity factor which varies over the range.

roscoe81, 2020-07-01:
I agree with you @MetPhoto, that simply recording and displaying raw resistance levels is of dubious value and it would be much better to use the MiCS-6814 data sheet to convert those levels to ppm concentrations.

It appears that you're proposing to map the "Sensing Resistance in air" specs for each sensor (on page 2 of the data sheet) against the gas concentration graphs for each sensor (on page 1 of the data sheet) by using the end-points and linear interpolation.

If that understanding is correct, I see a few issues, based on my understanding of how the MiCS-6814 works.

The main issue I see is that the "Sensing Resistance in air" specs are not the resistance values for the gas concentration extremities. They're the range of resistance values for each sensor in "synthetic air" (i.e. where the gases level are consistent with ambient air). Those resistance values do need to be measured as part of the calibration process and used to set R0 for each sensor. I therefore don't believe that you can use the "Sensing Resistance in air" specs in the way you propose.

The other issues are that the graphs on page 1 don't map raw resistance values against gas concentration levels. They map the raw resistance value divided by R0 (i.e. the resistance value in "air") against gas concentration levels. They're also logarithmic graphs and interp does linear interpolation.

I spent many hours working through the data sheet and I hope this explanation helps you. The exercise in converting raw resistance values into gas concentration levels is not easy and not exact. It requires calibration in "air" (for R0), compensation for temperature/humidity/air pressure/time and even then, each sensor responds to a broad spectrum of gases.

If you're interested, I have come up with a method here: https://github.com/roscoe81/enviro-monitor but it's still far from perfect.

MetPhoto, 2020-07-02:
Hi @roscoe81 I had series of similar thoughts.  Reporting of resistance values of little real use.  The use of 'synthetic air' in the specs also concerned me; how would this relate to world usage?

Finally the non-linearity of the graphs was also a concern. I was left wondering how these sensors are implemented in commercial applications.

Thanks for pointer to your project and the Adafruit IO example looks like a lot of fun.

closed 2023-01-20 by Gadgetoid