pimoroni/enviroplus-python #84: Running live vs cron

nikrawlinson, 2020-09-03:
I've tinkered with the weather.py code to create my own routine. I've noticed that when you read data from the sensors just once you get a different measurement to what you get if you loop through the process a few times as it seems to correct itself after the second reading. So I've set it to loop ten times, and then write the tenth set of readings to a CSV file.

If I run the routine at the terminal prompt (python /home/pi/enviro.py) it writes accurate readings to the CSV. However, I've set up a cron job to run it every 30 minutes, and always the readings it outputs are exactly the same, without even a millionth of a degree difference between each running:

Temperature: 22.54967373898835
Pressure: 676.4017617518247
Humidity: 89.2103699647858

Can anybody show me what I'm doing wrong?

`
#!/usr/bin/env python3

import time

from bme280 import BME280

try:
    from smbus2 import SMBus
except ImportError:
    from smbus import SMBus

import logging

logging.basicConfig(
    format='%(asctime)s.%(msecs)03d %(levelname)-8s %(message)s',
    level=logging.INFO,
    datefmt='%Y-%m-%d %H:%M:%S')

logging.info("""Print readings from the BME280 weather sensor.
""")

bus = SMBus(1)
bme280 = BME280(i2c_dev=bus)

from datetime import datetime

now = datetime.now()
thedate = now.strftime("%Y-%m-%d")
thetime = now.strftime("%H:%M:%S")

counter = 0

while counter < 10:
    temperature = bme280.get_temperature()
    pressure = bme280.get_pressure()
    humidity = bme280.get_humidity()
    logging.info("""Temperature: {:05.2f} *C
Pressure: {:05.2f} hPa
Relative humidity: {:05.2f} %
""".format(temperature, pressure, humidity))
    counter = counter + 1

import csv
csvRow = [thedate, thetime, temperature, pressure, humidity]
csvfile = "/home/pi/lounge.csv"
with open(csvfile, "a") as fp:
    wr = csv.writer(fp, dialect='excel')
    wr.writerow(csvRow)
`

Gadgetoid, 2020-09-03:
A pressure reading of 676 suggests you're either at high altitude, in the middle of a tornado, or something is going very, very wrong with reading data from the sensor. 90% relative humidity also strikes me as somewhat high. Of course you could just be somewhere decidedly more exciting than rainy northern England.

Trouble is, I don't have the first clue what could be wrong here. Do you have a picture of your setup, and details of how it's scheduled in crontab?

Presumably when you run it in the terminal, you see more realistic figures?

nikrawlinson, 2020-09-04:
Thanks Philip @Gadgetoid .

Sadly I'm not anywhere more exciting than rainy England. My set-up is as simple as can be: the Enviro+ connected to a Zero. I've attached a pic below.

Interestingly, I am running the same code on two identical devices - one upstairs and one downstairs and they both deliver identical results. Both work fine when invoking the script at the command prompt but resort to the values in my original post when trigged every half hour via cron.

The cron entry is:

*/30 * * * * python3 /home/pi/enviro4.py

On the off-chance it would make any difference (it didn't) I've also tried:

*/30 * * * * python /home/pi/enviro4.py

[image]

okslater, 2020-10-03:
The exact same thing is happening to me with an Enviro on an rPi Zero. I noticed also that the example programs weather.py and combined.py give different results when I run them from the keyboard. The humidity is never correct while the temperature is sometimes in the vicinity.

Gadgetoid, 2020-10-03:
Sorry for the delay replying- I've been torn between projects and unable to get a thought in edgeways.

This is strange, and I really need to find the time to figure out exactly what's going on.

I'd probably favor a Python daemon set up to run via a simple systemd unit than crontab. Though this would work-around and not *fix* the problem.

cameronbunce, 2020-12-17:
This issue caught my eye and I tried to replicate it myself
Pi Zero  - $ cat /proc/version 
Linux version 5.4.79+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 Mon Nov 23 13:18:15 GMT 2020
 brand new Enviro + phat,

I set a cron job under root to run your code */5 minutes and also ran it manually at odd intervals. Great Wall of text follows from the csv file, but I did see some interesting things. 

I would have expected that this was caused by some race condition with the device or permissions, but when I was running another of the examples scripts to pull info from every sensor, it didn't experience any trouble and instead got good values. Conversely, when I stepped away for lunch and ran errands it spit out 12 lines of the same values. looking up at the other values I see that 12:05 ran, and had good values, but 12:10 followed it and seemed to be the ( now shown to be ) junk values. 
I stopped the cron for a bit, looked around without suspecting anything, ran other programs and started the cron again. 

I can try for a cleaner test, but this to me seems like we aren't releasing the device or priming it or something like that. 

2020-12-17,11:58:47,20.909537659908892,993.7576566042007,31.34026509$
2020-12-17,12:05:02,21.428520247718552,993.7521019575927,30.55469157$
2020-12-17,12:10:02,22.302337288856506,660.3552733747258,84.01657101$
2020-12-17,12:44:06,23.047677277164098,993.5737507263558,27.67800860$
2020-12-17,12:45:02,22.302337288856506,660.3552733747258,84.01657101$
2020-12-17,12:45:56,23.277183920290554,993.5270494108938,27.36898427$
2020-12-17,12:50:03,25.394860911483192,993.5012021867859,24.66900911$
2020-12-17,12:55:03,27.460497636334935,993.4262806573248,21.77183399$
2020-12-17,13:00:03,27.210143535505722,993.4327780097856,21.92963537$
2020-12-17,13:05:01,22.302337288856506,660.3552733747258,84.01657101$
... 10 lines of the same ...
2020-12-17,14:00:02,22.302337288856506,660.3552733747258,84.01657101$
2020-12-17,14:05:03,24.607829187180528,993.3277111238572,25.42713763$
2020-12-17,14:10:01,22.302337288856506,660.3552733747258,84.01657101$
2020-12-17,14:15:02,22.302337288856506,660.3552733747258,84.01657101$

skillmatic-co, 2022-02-14:
Same thing is happening to me. Modified weather.py to create my own script that runs via cron every hour on the hour, and stores the values to a SQLite database. The issue is, my values are exactly the same every single time.

Gadgetoid, 2022-02-14:
Sorry I'm a total tool and missed the elephant in the room here.

Almost all environmental sensors take a moment to bed in and will likely send spurious initial readings. Running a script to interface with them via crontab is never going to produce reliable results.

The correct approach is - indeed - to write a Python script that polls the sensors in a loop and manage it with systemd like a service.

@cameronbunce pretty much nailed it- the sensor isn't being "primed". They have some warmup time, and tend to produce more reliable readings if kept running continuously.

We probably need a good example of how to do this.

@skillmatic-co's approach works because it gives the sensors time to settle, but it's still suboptimal. For best results They really should be run continuously and not periodically reset.

closed 2023-01-20 by Gadgetoid