pimoroni/enviroplus-python #87: Raspberry Pi 4 Enviro+ examples error.

mskuratowski, 2020-10-18:
Hi,

I'm using Raspberry Pi 4.
I'm getting an error when I try to run example weather.

[image]

Maybe I have a problem w Enviro+ because the display always blank.

[image]

Do you have any idea what could be wrong?

Gadgetoid, 2020-10-18:
Could you post a side view of the Enviro+ connected to your Pi? Both from the header side of the Pi (or the HDMI side) and the SD card side ideally.

mskuratowski, 2020-10-18:
@Gadgetoid 
Sure, below you can find images. If you need more, please let me know. 

[image]
[image]
[image]

Gadgetoid, 2020-10-18:
Connections look good! Nothing too alarming there. Is it possible the heatsink could have/could be contacting the bottom of the PCB and making something go awry?

mskuratowski, 2020-10-18:
@Gadgetoid Heatsink is contacted with an Enviro+ PM input.

Gadgetoid, 2020-10-19:
Sorry for the delay getting back- I can't see what's wrong here and suspect it might be a board fault. You should drop a line to support@pimoroni.com to see if they have any insight, otherwise they should be able to sort you out with a replacement.

mskuratowski, 2020-10-22:
No worries. I sent back enviro+ and I will order a new one. Thanks for your help.