pimoroni/enviroplus-python #90: Problems getting started

ajlennon, 2020-12-19:
Hi,

I'm following the getting started instructions [here](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-enviro-plus)

- I've installed and updated Raspian on my Pi Zero
- When I run the install script I get this

[image]

Can you advise?

ajlennon, 2020-12-19:
Looks like an issue with Python2 ?

ohnotnow, 2020-12-22:
I'm seeing this too - fresh install of raspbian and followed the same guide.  The install does complete eventually - just gives that error twice.  Most of the examples run ok - but the particulates.py crashes out with
```
Traceback (most recent call last):
  File "particulates.py", line 24, in <module>
    readings = pms5003.read()
  File "/usr/local/lib/python2.7/dist-packages/pms5003-0.0.5-py2.7.egg/pms5003/__init__.py", line 127, in read
    raise SerialTimeoutError("PMS5003 Read Timeout: Failed to read start of frame byte")
pms5003.SerialTimeoutError: PMS5003 Read Timeout: Failed to read start of frame byte
```
Not sure if the two are related though - first time using the enviro+ so not very familiar with debugging it.

ajlennon, 2020-12-22:
fwiw I had trouble with the particle sensor until I disabled the serial console

ohnotnow, 2020-12-22:
Hrm - if I disable the serial port I get an error saying it could not open the serial device.  Or did you mean leave the actual serial port enabled but disable logins via serial?

ajlennon, 2020-12-22:
Just the serial console as I think the output interferes with the particle sensor comms.

I did something like this

`# systemctl mask serial-getty@serial0.service`

ohnotnow, 2020-12-22:
Hrm - no difference I'm afraid, even after a reboot.  Also tried via the raspi-config and no difference there either.

ajlennon, 2020-12-22:
Humm. Not sure then I'm afraid. After I made that change my particulate sensor was working OK

ohnotnow, 2020-12-22:
Thanks for the suggestion anyway.  I'll see if anyone else chips in!

wgutmann, 2020-12-22:
I had the same thing happen today.
Forged on with thing anyway, ended up getting some "modules not found" errors, Fixed that by installing the modules but other errors surfaced as well. Were you able to get this resolved?

ohnotnow, 2020-12-23:
I just realised the enviro+ doesn't have the particulate sensor built in (yes, I'm thick).  So that would probably explain why it can't talk to it.  

I'll just get my coat....

ajlennon, 2020-12-23:
Ahahahaah. Nice one! Thanks for holding your hand up to that you gave me a laugh!

Sounds like something I would do!

ajlennon, 2020-12-23:
> I had the same thing happen today.
> Forged on with thing anyway, ended up getting some "modules not found" errors, Fixed that by installing the modules but other errors surfaced as well. Were you able to get this resolved?

Yes it's all working now -

https://github.com/DynamicDevices/enviroplus-docker-image

maceh, 2021-01-19:
I've just gone through the steps and noticed this error message pop up in the process. However I just left my pi for a few mins and it appeared to have skipped over this error and continued. Not actually sure if this sounddevice.py matters at all for this board?

I've only run through each of the example python scripts so far, but all of them appear to work (aside from the particulate sensor one, since the board doesn't have one).

If you're running the all-in-one example without a PM sensor, use ./all-in-one-no-pm.py
Also touch the light sensor to cycle between the different sensors..

ajlennon, 2021-01-19:
fwiw I came a cropper because I bought the board variant without the gas sensor on it (my mistake - I got confused).

So because that's not present some of the examples fail. fwiw it might be worth trapping that and gracefully working around absence of the gas sensor on that board variant as it's a bit confusing otherwise.

closed 2023-01-20 by Gadgetoid