pimoroni/enviroplus-python #91: No display for noise analysis example

ajlennon, 2020-12-19:
When I run `python3 lcd.py` I get the expected output on the display

[image]

However when I run `python3 noise-profile.py` I see the backlight turn on but there's no display?

[image]

rezzorix, 2021-01-18:
Probably has to do with this: https://github.com/pimoroni/get/issues/209

Gadgetoid, 2022-02-11:
What OS/revision are you using?

I'm seeing this on Raspberry Pi OS 64bit on a Pi 4. The low, mid and high frequencies are all reading as zero, which causes nothing to display. No amount of shouting at my Enviro+ board will register any kind of change.

I'm guessing the introduction of Pulse Audio as standard, or something else changing since I wrote this code, has broken the i2s audio input.

Looks like the following successfully captures audio:

```
arecord -f S32_LE -c 2 -r 16000 -D hw:adau7002,0 > test-s32_le.wav
```

And this diff gets Enviro+ noise support working again for me:

```
diff --git a/library/enviroplus/noise.py b/library/enviroplus/noise.py
index 7b6d5e2..0d413bd 100644
--- a/library/enviroplus/noise.py
+++ b/library/enviroplus/noise.py
@@ -83,6 +83,7 @@ class Noise():
     def _record(self):
         return sounddevice.rec(
             int(self.duration * self.sample_rate),
+            device='adau7002',
             samplerate=self.sample_rate,
             blocking=True,
             channels=1,
```

Sooo it looks like it was trying to capture from the default device.

closed 2023-01-20 by Gadgetoid

nick-dirienzo, 2024-02-04:
Hi there,

I ran into the same problem as @ajlennon and after looking through your thoughts (@Gadgetoid), I still can't determine what the problem is. Has anyone found a way to solve the issue of getting the python noise-profile.py results to display on the LCD display of the Enviro+ hat?

Thanks,

Nick