pimoroni/enviroplus-python #97: LCD does not show anything only backlight is on

fotisK69, 2021-02-17:
Hi,

after I have installed the enviro+ and update/upgrade my pi zero W i tried the example of lcd.py and the only thing that happens is that the backlight is on but no "Hello" message!

How can i solve this, it seems that all the connections are fine since i do get measurements when executing other example files.

spanner3003, 2021-02-20:
Hi I have this same problem and believe it is something to do with the most recent kernel update on the pi4 for me.

Gadgetoid, 2022-02-11:
At first I thought I could replicate this, but then I noticed two missing placements on my Enviro+ which made the screen unhappy.

I have display output on the latest 64bit Raspberry Pi OS on a Pi 4, so perhaps it's been fixed since? Or was some other complicating factor.

Let me know if you're still having issues... I know it's been a while, sorry!

closed 2022-02-11 by Gadgetoid