pimoroni/enviroplus-python #103: Install script does not install all required python libraries

hardfemmegarage, 2021-09-28:
Ran the install script. Could not run examples out of the box, even though that was the next step in the tutorial. I needed to manual install bme280.

Still need to locate ltr559 and ads1015, wherever they are.

There is nothing about this in the tutorial and the implication is that the library should be complete. Get it together and fix this, please or at least make a note of this in your tutorial. Not everyone has the time or skills to go looking for the required libraries. Even those who are skilled just want to install the libraries and use the hardware without having to hunt it down. It's aggravating.

keyboard-magpie, 2021-10-02:
Works fine on a fresh install of raspbian for me with one exception:

`modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.17+/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.10.17+
`

Just testing if `sudo rpi-update` fixes this.

keyboard-magpie, 2021-10-02:
Can confirm `sudo rpi-update` fixed my issues.

Gadgetoid, 2021-11-02:
`pimoroni-bme280`, `ltr559` and `ads1015` are all listed as dependencies, they should be installed automatically when the library is installed.

Note: don't run `sudo rpi-update`, the RPi kernel/modules/firmware gets updated with a regular `sudo apt update && sudo apt upgrade` whereas `rpi-update` is just a means for installing (potentially very broken) pre-release versions.

Gadgetoid, 2021-11-04:
Now added a new `check-install.py` to help diagnose these issues: https://github.com/pimoroni/enviroplus-python/blob/master/check-install.py

Run with:

```
python3 check-install.py
```

closed 2023-01-20 by Gadgetoid