pimoroni/enviroplus-python #105: Gas sensors not returning values

timiman, 2021-10-21:
Having installed all the required packages and working just fine for two days,
I'm started getting the error below when trying to gas.read_all().

Sample code I use:

```
#!/usr/bin/env python3
from enviroplus import gas
try:
    while True:
        try:
            # Get gas readings
            readings = gas.read_all()
            oxid = readings.oxidising
            oxid = "{:.2f}".format(oxid)
            reduc = readings.reducing
            reduc = "{:.2f}".format(reduc)
            nh3 = readings.nh3
            nh3 = "{:.2f}".format(nh3)
            readingCounter+=1

            if readingCounter > warmupreadings:
                #Send to DB (enviroment table)

            time.sleep(timeout)
        except Exception as exc:
            time.sleep(timeout)
            logging.info('Generic Exception 1: ', exc)

except Exception as exc:
    pass

```

The error is:

--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.7-py3.7.egg/i2cdevice/__init__.py", line 250, in get_field
    value = field.adapter._decode(value)
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.7-py3.7.egg/i2cdevice/adapter.py", line 26, in _decode
    raise ValueError("{} not in lookup table".format(value))
ValueError: 7 not in lookup table

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gas.py", line 32, in <module>
    readings = gas.read_all()
  File "/usr/local/lib/python3.7/dist-packages/enviroplus-0.0.4-py3.7.egg/enviroplus/gas.py", line 83, in read_all
    setup()
  File "/usr/local/lib/python3.7/dist-packages/enviroplus-0.0.4-py3.7.egg/enviroplus/gas.py", line 50, in setup
    adc_type = adc.detect_chip_type()
  File "/usr/local/lib/python3.7/dist-packages/ads1015-0.0.7-py3.7.egg/ads1015/__init__.py", line 170, in detect_chip_type
    while self._ads1015.get('CONFIG').operational_status == 'active':
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.7-py3.7.egg/i2cdevice/__init__.py", line 233, in get
    result[field] = self.get_field(register, field)
  File "/usr/local/lib/python3.7/dist-packages/i2cdevice-0.0.7-py3.7.egg/i2cdevice/__init__.py", line 252, in get_field
    raise ValueError("{}: {}".format(field.name, str(value_error)))
ValueError: data_rate_sps: 7 not in lookup table

The other sensors of Enviro + are working just fine.
Is it a hardware error or a python3 error ?

Gadgetoid, 2021-11-02:
> Could it be a compatibility issue ?

It's possible the right version of the ads1015 library wasn't installed, though it should have been installed automatically with the library.

You're not the first person to have missing/broken dependencies, though, how did you install the library?

timiman, 2021-11-03:
> You're not the first person to have missing/broken dependencies, though, how did you install the library?

Both times I used the:
    git clone https://github.com/pimoroni/enviroplus-python
    cd enviroplus-python
    sudo ./install.sh

Up to now, sensors readings are fine.

Gadgetoid, 2021-11-03:
Thank you - I might have to run a few tests to make sure that installer isn't breaking in some weird way.

closed 2023-01-20 by Gadgetoid