pimoroni/enviroplus-python #108: Install script is missing a of dependencies and simply does not work

Kareltje1980, 2021-11-03:
Board: RPI4 4GB
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

Fresh install, including an ```apt update``` and ```apt upgrade```. 
git cloned the repository and performed ./install.sh as per tutorial

All fine up till now. 

But then when trying to run  all-in-one-no-pm.py

## Error 1
```
Traceback (most recent call last):
  File "all-in-one-no-pm.py", line 7, in <module>
    import ST7735
ImportError: No module named ST7735
```
fixed using: pip install ST7735

## Error 2
```
Traceback (most recent call last):
  File "all-in-one-no-pm.py", line 13, in <module>
    import ltr559
ImportError: No module named ltr559
```
fixed using: pip install ltr559

## Error 3
```
Traceback (most recent call last):
  File "all-in-one-no-pm.py", line 11, in <module>
    ltr559 = LTR559()
  File "/home/pi/.local/lib/python2.7/site-packages/ltr559/__init__.py", line 212, in __init__
    BitField('ALS', 0x0F)
  File "/home/pi/.local/lib/python2.7/site-packages/i2cdevice/__init__.py", line 166, in __init__
    self._i2c = smbus.SMBus(1)
IOError: [Errno 2] No such file or directory
```

And I am out of ideas at this point...

Gadgetoid, 2021-11-04:
The installer now only installs for python3. You should run examples with Python 3.

Failing to set up the i2s bus is an issue, though, the script should be doing that. Did you see any error output when running the installer? What's the content of your `/boot/config.txt`

Gadgetoid, 2021-11-04:
Grab and run `check-install.py` to verify everything is where it should be - https://github.com/pimoroni/enviroplus-python/blob/master/check-install.py

Run with:

```
python3 check-install.py
```

Kareltje1980, 2021-11-04:
**Thanks!!** The python3 did the trick. This is the config.txt

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1
dtoverlay=pi3-miniuart-bt
dtoverlay=adau7002-simple
```

The install script confirms all is working. 

```
Checking Enviro+ install, please wait...

System dependencies...
  Retrieving cache...
  Checking for python3-rpi.gpio    ✅
  Checking for python3-cffi        ✅
  Checking for python3-smbus       ✅
  Checking for python3-numpy       ✅
  Checking for python3             ✅
  Checking for python3-pip         ✅
  Checking for python3-pil         ✅
  Checking for python3-spidev      ✅
  Checking for libportaudio2       ✅

Python dependencies...
  Checking for bme280              ✅
  Checking for pms5003             ✅
  Checking for ltr559              ✅
  Checking for ST7735              ✅
  Checking for ads1015             ✅
  Checking for fonts               ✅
  Checking for font_roboto         ✅
  Checking for astral              ✅
  Checking for pytz                ✅
  Checking for sounddevice         ✅
  Checking for paho.mqtt           ✅

System config...
  Checking for dtparam=spi=on in /boot/config.txt: ✅
  Checking for dtoverlay=adau7002-simple in /boot/config.txt: ✅
  Checking for enable_uart=1 in /boot/config.txt: ✅
  Checking for dtparam=i2c_arm=on in /boot/config.txt: ✅
  Checking for dtoverlay=pi3-miniuart-bt in /boot/config.txt: ✅

✅  Looks good from here!

Have you?
  • Rebooted after installing
  • Made sure to run examples with "python3"
  • Checked for any errors when running "sudo ./install.sh"
```

closed 2021-11-04 by Kareltje1980