pimoroni/enviroplus-python #111: PMS5003 best use

liviuncl, 2022-02-11:
I'd like to understand how should I read the values from the PMS5003, because I have the feeling something is wrong with what I'm doing when reading the values from the PMS5003

For example, if I use the pms5003.read() in a loop, every 30 seconds, I get values that stay in the same range for a long period of time. For the last 2 days, the values for PM2.5 and PM10 stayed in 3-8 and 4-11 range respectively, they were not completely constant to draw your attention, but also were not the "real" values.

When I connected to the RPi and run the particulates.py sample code, the values were "updated" in a new range, jumping to 34 and 49 respectively (the values were validated by another station nearby) but the behaviour stays the same .. they will remain in the those ranges again for a long period.

It's like the sensor is "sleeping" and only giving some values in the same interval forever.
Running the particulates.py has an effect on gas values - reseting them as well - like a wake up call.

liviuncl, 2022-02-12:
I'm adding some info, maybe that could lead to some more answers.
When running the sample code particulates.py, the first 20=30 readings are not stable..like it performs the calibration, but after those readings the values stabilise and are not longer complete. Can I trust the values I get after the first 30-40 readings? 

[image]

roscoe81, 2022-02-12:
@liviuncl  I suspect that the issue is that you're taking a reading from the PMS5003 every 30 seconds and its buffer is full. Try reading it in a loop that has no delay. If you only want to capture a reading every 30 seconds, you can do that in your code but continue to take the data from the PMS5003 continuously in between that capture time.

liviuncl, 2022-02-12:
Thanks very much Ross, that makes sense - I've thought that adding a sleep might help with two aspects 
- too much data with second granularity doesn't help me much (I can deal with that in a different way)
- wearing down the PMS5003, even though the sleep doesn't have any effect I presume on the laser/fan of the sensor, so actually that doesn't help 

How do you deal with the PMS5003 limited lifespan?

roscoe81, 2022-02-12:
I suspect that it won’t “wear down” the PMS5003 @liviuncl and I’m not sure that you have a choice. From my experience, having  a delay in your loop won’t work. I’ve been running multiple PMS5003 units with continuous loops for two years without problems.

liviuncl, 2022-02-12:
@roscoe81 thanks so much, it gives me hope and some room to continue playing with the enviro+.

closed 2022-02-12 by liviuncl

Gadgetoid, 2022-02-13:
Normally the PMS5003 sends measurements every whether you want it to or not, failing to read them just results in the UART buffer (on the Raspberry Pi side) filling up until it overflows. This used to break the library, since it'd corrupt the reading(s) on the overflow boundary.

Thanks @roscoe81 I would not have realised this!

Additionally during high rates of data change the PMS5003 will switch into "fast mode" and can send updates every 200ms. This exacerbates the backlog problem.

I have an open issue reminding me to do something about this, but threading is perhaps not the correct answer. https://github.com/pimoroni/pms5003-python/issues/6

As mentioned in this issue, implementing "passive" mode is probably the solution. More detail here: https://github.com/pimoroni/pms5003-circuitpython/issues/5

I don't believe passive mode has any implications for lifespan, since the sensor must run continuously for reliable readings, but it potentially makes handling updates in code a lot easier.