pimoroni/enviroplus-python #115: light.py sample fails to run on new Enviro+

pjsgsy, 2022-07-11:
Just reiceived a new Enviro+. Most of the sample seem to run, so the module is working, but the light.py sample does not. It produces this error on startup

:~/Pimoroni/enviroplus/examples $ pi@x1:~/Pimoroni/enviroplus/examples $ python3 light.py
Traceback (most recent call last):
  File "/home/pi/Pimoroni/enviroplus/examples/light.py", line 7, in <module>
    from ltr559 import LTR559
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 786, in exec_module
  File "<frozen importlib._bootstrap_external>", line 918, in get_code
  File "<frozen importlib._bootstrap_external>", line 587, in _compile_bytecode
ValueError: bad marshal data (unknown type code)
ValueError: bad marshal data (unknown type code)

i2c seems enabled and the 23, which I belive is the sensor is there 

1:~/Pimoroni/enviroplus/examples $  sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- 23 -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

Any ideas what I am doing wrong? Pi is up to date (and fresh) install of NOOBS. I used the one-line install for the samples.

Thanks.

andypiper, 2022-08-17:
Weird - what does `pip list` show - is `ltr559` in there?

This is working for me, but it is some time since I installed things...

Gadgetoid, 2022-08-23:
The `ValueError: bad marshal data (unknown type code)` seems to be the crux of the issue?

I've not come across this before, but it suggests corruption in the `.pyc` files and might suggest a wider system issue.

Try reinstalling the library with `sudo python3 -m pip install --upgrade --force-reinstall ltr559`

closed 2024-04-25 by helgibbons