pimoroni/enviroplus-python #127: install.sh not working when trying to install over SSH

highfly117, 2023-10-31:
I'm trying to get my enviro+ set up but the install.sh keeps giving an error that my enviro externally-managed-environment 

I tried setting up a virtual environment and then running the install.sh but i get the same error message

hulleyrob, 2023-10-31:
Did something not paste properly into the comment as it seems like something is missing?

Gadgetoid, 2023-10-31:
This is happening because of changes to Raspberry Pi OS, I'm in the process of fixing our code but... there's quite a lot of it. In particular I am trying to focus on Enviro right now, so expect things to start working... soooon.

The what and why of this chaos is explained somewhat here: https://github.com/pimoroni/boilerplate-python/pull/13

Progress on the fixes is detailed here: https://github.com/pimoroni/enviroplus-python/pull/126

highfly117, 2023-10-31:
so can we image an older version of raspberry pi OS and if so whats the latest version we can use?

Gadgetoid, 2023-10-31:
> so can we image an older version of raspberry pi OS and if so whats the latest version we can use?

On a Pi 5. no.

On a Pi 5. Sure- 
- arm64 (64bit) - http://downloads.raspberrypi.org/raspios_oldstable_arm64/images/raspios_oldstable_arm64-2023-10-10/2023-05-03-raspios-bullseye-arm64.img.xz
- armhf (32bit) - http://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2023-10-10/2023-05-03-raspios-bullseye-armhf.img.xz

peardox, 2023-11-10:
Just got my Enviro+ and it immediately fails to work cos the instructions are wrong (unamused)

A little digging and a hacky solution is

> pip install enviroplus --break-system-packages

Now examples work

helgibbons, 2024-04-25:
The installation instructions on Github (and [on Learn](https://learn.pimoroni.com/article/getting-started-with-enviro-plus)) should now be up to date (and hopefully more helpful with regards to virtual environments). Sorry that things have been a little chaotic for the last few months!

Closing this one for now, but let us know if you're still having issues.

closed 2024-04-25 by helgibbons