pimoroni/enviroplus-python #128: Installing enviroplus-python library and running lcd.py does not work

gardiner, 2023-11-18:
After installing enviroplus-python library as described in the documentation the `lcd.py` example fails with an error `'int' object is not iterable'`.

The reason is that the used [ST7735](https://github.com/pimoroni/st7735-python) library has had a breaking change (see https://github.com/pimoroni/st7735-python/issues/32 for details).

Until the code in this repository is fixed you can solve the problem like this:

* uninstall ST7735 library (e.g. `pip uninstall st7735`)
* clone ST7735 repository (e.g. `git clone https://github.com/pimoroni/st7735-python.git`)
* checkout commit before introducing breaking change (e.g. `git checkout c0733d6`)
* install *that* version (e.g. `pip install library/`)

Note: If you are using a newer Pi OS using python virtual environments, you have to remember that `su -` and `sudo` do not respect a previously selected venv, so whenever you switch between the normal user and root, you have to make sure you activate the venv again.

Gadgetoid, 2023-11-21:
The updated examples are in an open pull request, you should use the library and examples from there until it's released/merged... probably something I should just do ASAP, but I'd hoped to get some feedback on it first - 

See: https://github.com/pimoroni/enviroplus-python/pull/126

helgibbons, 2024-04-25:
Closing this one now that these changes are released/merged.

closed 2024-04-25 by helgibbons