pimoroni/enviroplus-python #132: Will not install. Tried multiple methods and multiple pi's and multiple pi os versions.

cjo40, 2024-04-21:
Will not install. Tried "one-line" and " Install and configure" and neither works on rpi4 on bullseye or bookworm. Bash and directions need updated.

Gadgetoid, 2024-04-22:
Oof, I missed updating the readme. Yes install should be:

```
git clone https://github.com/pimoroni/enviroplus-python
cd enviroplus-python
./install.sh
```

cjo40, 2024-04-23:
cjo40@raspberrypi:~/enviroplus-python $ ./install.sh
Installing enviroplus...

This script should be run in a virtual Python environment.
Would you like us to create and/or use a default one? [y/N] y

Creating a new virtual Python environment in /home/cjo40/.virtualenvs/pimoroni, please wait...

Checking for /home/cjo40/Pimoroni/auto_venv.sh

Creating /home/cjo40/Pimoroni/auto_venv.sh

Checking Dependencies. Please wait...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: toml in /usr/lib/python3/dist-packages (0.10.1)
Collecting toml
  Downloading https://www.piwheels.org/simple/toml/toml-0.10.2-py2.py3-none-any.whl (16 kB)
Installing collected packages: toml
  Attempting uninstall: toml
    Found existing installation: toml 0.10.1
    Not uninstalling toml at /usr/lib/python3/dist-packages, outside environment /home/cjo40/.virtualenvs/pimoroni
    Can't uninstall 'toml'. No files were found to uninstall.
Successfully installed toml-0.10.2

Installing for Python 3.9.2...

Checking for python3
Checking for python3-pip
Checking for python3-cffi
Checking for libportaudio2

Installing missing packages: python3-cffi libportaudio2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  python3-ply python3-pycparser
Suggested packages:
  python-ply-doc
The following NEW packages will be installed:
  libportaudio2 python3-cffi python3-ply python3-pycparser
0 upgraded, 4 newly installed, 0 to remove and 21 not upgraded.
Need to get 289 kB of archives.
After this operation, 1,475 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 libportaudio2 arm64 19.6.0-1.1 [61.3 kB]
Get:2 http://deb.debian.org/debian bullseye/main arm64 python3-ply all 3.11-4 [65.5 kB]
Get:3 http://deb.debian.org/debian bullseye/main arm64 python3-pycparser all 2.20-3 [74.5 kB]
Get:4 http://deb.debian.org/debian bullseye/main arm64 python3-cffi all 1.14.5-1 [87.9 kB]
Fetched 289 kB in 0s (2,338 kB/s)  
Selecting previously unselected package libportaudio2:arm64.
(Reading database ... 157535 files and directories currently installed.)
Preparing to unpack .../libportaudio2_19.6.0-1.1_arm64.deb ...
Unpacking libportaudio2:arm64 (19.6.0-1.1) ...
Selecting previously unselected package python3-ply.
Preparing to unpack .../python3-ply_3.11-4_all.deb ...
Unpacking python3-ply (3.11-4) ...
Selecting previously unselected package python3-pycparser.
Preparing to unpack .../python3-pycparser_2.20-3_all.deb ...
Unpacking python3-pycparser (2.20-3) ...
Selecting previously unselected package python3-cffi.
Preparing to unpack .../python3-cffi_1.14.5-1_all.deb ...
Unpacking python3-cffi (1.14.5-1) ...
Setting up libportaudio2:arm64 (19.6.0-1.1) ...
Setting up python3-ply (3.11-4) ...
Setting up python3-pycparser (2.20-3) ...
Setting up python3-cffi (1.14.5-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u8) ...
[... 161 lines cut ...]
Installing collected packages: smbus2, gpiod, i2cdevice, gpiodevice, st7735, sounddevice, pytz, pms5003, pimoroni-bme280, paho-mqtt, ltr559, fonts, font-roboto, astral, ads1015, enviroplus
Successfully installed ads1015-1.0.0 astral-3.2 enviroplus-0.0.6 font-roboto-0.0.1 fonts-0.0.3 gpiod-1.5.4 gpiodevice-0.0.3 i2cdevice-1.0.0 ltr559-1.0.0 paho-mqtt-2.0.0 pimoroni-bme280-1.0.0 pms5003-1.0.0 pytz-2024.1 smbus2-0.4.3 sounddevice-0.4.6 st7735-1.0.0
Done!

Using config.txt in /boot

Running setup commands...

Setting up i2c and SPI..
Backing up /boot/config.txt to /boot/config.preinstall-enviroplus-2024-04-22-19-00-57.txt

Running: "sudo raspi-config nonint do_spi 0"
Running: "sudo raspi-config nonint do_i2c 0"
Setting up serial for PMS5003..
Running: "sudo raspi-config nonint do_serial_cons 1"
/usr/bin/raspi-config: 2909: do_serial_cons: not found
⚠ WARNING: ^^^ 😬 previous command did not exit cleanly!
Running: "sudo raspi-config nonint do_serial_hw 1"
/usr/bin/raspi-config: 2909: do_serial_hw: not found
⚠ WARNING: ^^^ 😬 previous command did not exit cleanly!

Adding dtoverlay=pi3-miniuart-bt to /boot/config.txt
dtoverlay=pi3-miniuart-bt 
Adding dtoverlay=adau7002-simple to /boot/config.txt
dtoverlay=adau7002-simple

cjo40, 2024-04-23:
Several errors during $ ./install.sh and none of the examples will work.

Gadgetoid, 2024-04-23:
Ooof, 100% of that wall of error seems to be related to gpiod builds. Though it appears to have been successfully installed anyway.

What's happening when you run the examples?

Note, you'll get dumped out of the virtualenv when the install finishes. To run the examples you'll need to:

```
source ~/.virtualenvs/pimoroni/bin/activate
```

I, uh, should probably document this somewhere!

It also helps if you:

```
sudo apt install python3-virtualenvwrapper virtualenvwrapper
```

Since then you can use the `workon` command to list envs and switch, eg to list:

```
phil@pi5-nvme:~ $ workon
decky
dev
lgpio
nvme
pimoroni
poetry
```

to switch:

```
phil@pi5-nvme:~ $ workon pimoroni
(pimoroni) phil@pi5-nvme:~ $
```

This is all assuming the virtual environment warts are the problem.

helgibbons, 2024-04-25:
The installation instructions on Learn are now up to date: https://learn.pimoroni.com/article/getting-started-with-enviro-plus

Get-Kraken, 2024-05-09:
> The installation instructions on Learn are now up to date: https://learn.pimoroni.com/article/getting-started-with-enviro-plus

Is this valid for the non plus version of the board? the enviro mini

helgibbons, 2024-05-10:
> Is this valid for the non plus version of the board? the enviro mini

[This one?](https://shop.pimoroni.com/products/enviro?variant=31155658489939) 

Yes! Both boards use the same installer (though you won't be able to run the examples that use the gas sensor or the particulate sensor on the non-plus version).

Gadgetoid, 2025-03-18:
Closing for now, much has changed/updated in the interim so please open a new issue or consult your nearest nerd should symptoms persist.