pimoroni/enviroplus-python #137: GPIO24: not found

cmsimike, 2024-04-23:
After the fix of #134 I tried re-installing and rerunning. I get passed the `get_pin` part now, but am getting `GPIO24: not found`

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/src/enviroplus-mqtt/src/logger.py", line 60, in __read_pms_continuously
    pms = PMS5003()
  File "/usr/local/lib/python3.9/site-packages/pms5003/__init__.py", line 109, in __init__
    self._pin_enable, self._pin_reset = gpiodevice.get_pins_for_platform(PLATFORMS)
ValueError: not enough values to unpack (expected 2, got 0)
Woah there, suitable gpiochip not found!
  ❌  GPIO24: not found - /dev/gpiochip2 (raspberrypi-exp-gpio)!
  ❌  GPIO24: not found - /dev/gpiochip1 (brcmvirt-gpio)!
  ❌  GPIO24: not found - /dev/gpiochip0 (pinctrl-bcm2835)!
```

This is running with `--use-pms5003` but even without that flag, I don't get the exception but do get the GPIO24: not found messages.

cmsimike, 2024-04-23:
I don't believe I have `gpioinfo` installed:
`sudo find / -name gpioinfo` finds nothing.

`{'RELEASE': '10', 'CODENAME': 'buster', 'DESCRIPTION': 'Raspbian GNU/Linux 10 (buster)', 'ID': 'Raspbian'}
[]`

Gadgetoid, 2024-04-23:
Try:

```
sudo apt install gpiod
```

Which should give you this tool.

cmsimike, 2024-04-23:
Thanks! That worked. Here's the output:

```bash
piochip0 - 54 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed   "spi0 CS1"  output   active-low [used]
	line   8:      unnamed   "spi0 CS0"  output   active-low [used]
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       unused   input  active-high 
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       unused  output  active-high 
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused   input  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 
gpiochip1 - 2 lines:
	line   0:      unnamed       "led0"  output  active-high [used]
	line   1:      unnamed       unused   input  active-high 
gpiochip2 - 8 lines:
	line   0:      unnamed       unused  output  active-high 
	line   1:      unnamed       unused  output  active-high 
	line   2:      unnamed       unused  output  active-high 
	line   3:      unnamed       unused  output  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused  output  active-high 
	line   6:      unnamed       unused  output  active-high 
	line   7:      unnamed       "led1"   input  active-high [used]
```

Gadgetoid, 2024-04-23:
`unnamed` 💀 

Well that's... extremely unhelpful.

The library currently requires GPIO line labels - available in later versions of Raspberry Pi OS - in order to figure out which pin to use. Without this you've got to be very explicit about which GPIO device and line(s) you want to open, eg:

⚠️ the below example needs an experimental version of the PMS5003 library: 

```
pip install git+https://github.com/pimoroni/pms5003-python@patch-user-gpio
```

```python
#!/usr/bin/env python3

import logging
import time
from gpiod import Chip

from pms5003 import PMS5003, ReadTimeoutError, OUTL, OUTH

logging.basicConfig(
    format="%(asctime)s.%(msecs)03d %(levelname)-8s %(message)s",
    level=logging.INFO,
    datefmt="%Y-%m-%d %H:%M:%S")

logging.info("""particulates.py - Print readings from the PMS5003 particulate sensor.

Press Ctrl+C to exit!

""")

# Open the GPIO chip and request the enable/reset lines
chip = Chip("/dev/gpiochip0")
lines = chip.request_lines(consumer="pms5003", config={
    22: OUTH,
    27: OUTL
})

# Pass the requested lines and offsets into PMS5003...
pms5003 = PMS5003(pin_enable=(lines, 22), pin_reset=(lines, 27))
time.sleep(1.0)

try:
    while True:
        try:
            readings = pms5003.read()
            logging.info(readings)
        except ReadTimeoutError:
            pms5003 = PMS5003()
except KeyboardInterrupt:
    pass
```

cmsimike, 2024-04-28:
Sorry for the delay. I did what you said above:
```bash
pi@pi001:~ $ sudo python3 pyfoo.py
2024-04-28 16:47:04.535 INFO     particulates.py - Print readings from the PMS5003 particulate sensor.

Press Ctrl+C to exit!

Traceback (most recent call last):
  File "/home/pi/pyfoo.py", line 28, in <module>
    pms5003 = PMS5003(pin_enable=(lines, 22), pin_reset=(lines, 27))
  File "/usr/local/lib/python3.9/site-packages/pms5003/__init__.py", line 106, in __init__
    self._pin_enable = gpiodevice.get_pin(pin_enable, "PMS5003_en", OUTH)
  File "/usr/local/lib/python3.9/site-packages/gpiodevice/__init__.py", line 150, in get_pin
    chip = find_chip_by_pins(pin)
  File "/usr/local/lib/python3.9/site-packages/gpiodevice/errors.py", line 45, in wrapper
    errors.append(next(i))
  File "/usr/local/lib/python3.9/site-packages/gpiodevice/__init__.py", line 122, in find_chip_by_pins
    offset = chip.line_offset_from_id(pin_id)
  File "/usr/local/lib/python3.9/site-packages/gpiod/chip.py", line 131, in line_offset_from_id
    return self._chip.line_offset_from_id(id)
TypeError: argument 1 must be str, not LineRequest
```

Would it be easier if I just upgraded my OS?

cmsimike, 2024-05-12:
I just re-installed and am on Rasbian 12 and this works correctly again. Thanks!

closed 2024-05-12 by cmsimike