pimoroni/enviroplus-python #142: pms5003 on bookworm - Raspberry Pi 2

discosilvia, 2024-06-01:
Hi all,

I've followed both the automatic and manual install instructions, including the raspi-config settings, but the PMS5003 is still not working. It worked before on bullseye, but I had to do a fresh reinstall due to a faulty SD card.

Another troubleshooting fix I tried is downgrading from **pms5003 1.0.1 to 0.0.5,** but no luck.

when running particulates.py I get the following error:
`Woah there, suitable gpiochip not found!
  ✅  GPIO22: (line 22) found - /dev/gpiochip0 (pinctrl-bcm2835)!
  ⚠   GPIO22: (line 22, GPIO22) currently claimed by particulates-PMS5003_en
`

combined.py gives me `WARNING  Failed to read PMS5003`

sensorcommunity.py gives me 

``` File "/home/pi/enviroplus-python/examples/sensorcommunity.py", line 66, in read_values
    pm_values = pms5003.read()
                ^^^^^^^^^^^^^^
  File "/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/pms5003/__init__.py", line 132, in read
    raise ReadTimeoutError("PMS5003 Read Timeout: Could not find start of frame")
pms5003.ReadTimeoutError: PMS5003 Read Timeout: Could not find start of frame

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/enviroplus-python/examples/sensorcommunity.py", line 210, in <module>
    values = read_values()
             ^^^^^^^^^^^^^
  File "/home/pi/enviroplus-python/examples/sensorcommunity.py", line 72, in read_values
    pm_values = pms5003.read()
                ^^^^^^^^^^^^^^
  File "/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/pms5003/__init__.py", line 134, in read
    sof = self._serial.read(1)
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```
This is the output of my `pip freeze`:

```ads1015==1.0.0
astral==3.2
certifi==2022.9.24
cffi==1.15.1
chardet==5.1.0
charset-normalizer==3.0.1
colorzero==2.0
distro==1.8.0
enviroplus==1.0.2
font-roboto==0.0.1
fonts==0.0.3
gpiod==2.1.3
gpiodevice==0.0.5
gpiozero==2.0
i2cdevice==1.0.0
idna==3.3
lgpio==0.2.2.0
ltr559==1.0.0
numpy==1.24.2
olefile==0.46
paho-mqtt==2.1.0
pigpio==1.78
Pillow==9.4.0
pimoroni-bme280==1.0.0
ply==3.11
pms5003==1.0.1
pycparser==2.21
pycryptodomex==3.11.0
pyserial==3.5
python-apt==2.6.0
pytz==2024.1
requests==2.28.1
RPi.GPIO==0.7.1a4
six==1.16.0
smbus2==0.4.2
sounddevice==0.4.7
spidev==3.5
ssh-import-id==5.10
st7735==1.0.0
toml==0.10.2
urllib3==1.26.12
```

Thanks for the help and all your work!

Gadgetoid, 2024-06-02:
Do you have some code set to run on startup or otherwise? That pin should not be claimed unless another example is trying to talk to PMS5003- and that would certainly explain why UART/serial communications are going wobbly.

discosilvia, 2024-06-03:
Facepalm! It was indeed a process of a previously installed service using PMS5003. Once disable everything worked as expected

Thank you so much!

closed 2024-06-03 by discosilvia

Gadgetoid, 2024-06-03:
Whew great news! 🥳, I guess gpiod is useful for *something* 😆