pimoroni/enviroplus-python #145: Library for Display missing

torstenandrecht, 2024-08-03:
when I execute python all-in-one.py I get the following error message:

Traceback (most recent call last):
  File "/home/andrecht/enviroplus-python/examples/all-in-one.py", line 20, in <module>
    from PIL import Image, ImageDraw, ImageFont
ModuleNotFoundError: No module named 'PIL'

Solution:
pip install Pillow

Source: 
https://stackoverflow.com/questions/8863917/importerror-no-module-named-pil

slokhorst, 2024-08-17:
Same issue here. I also had to `apt install libopenblas0` on Raspberry Pi OS Lite Bookwork 32-bit to get the display to work.

arijitsadhu, 2024-12-02:
pip install Pillow failed to build the wheel, giving the error:
"RequiredDependencyException:
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source."

So following the link installed directly from apt:
sudo apt install python3-pil

gilbertfrancois, 2024-12-30:
I saw the same issue. The easiest way to fix it is to install both `numpy` and `pillow` system wide with `apt`. Then the required C libraries are included and the 2 python packages are also available in the virtual environment. 

```sh
sudo apt install python3-numpy python3-pil

```
However, in my case, the display still stays blank, without returning an error. I'll make a new issue for that.

Gadgetoid, 2025-03-18:
I've updated the installer to use `requirements-examples.txt` which now includes `pillow`. Hopefully fixing this issue.

As for the other comments-

Afaict `pillow==11.1.0` does not require `libopenblas0`. Just worked out of the box for me (piwheels probably vendors it now).

`numpy` is specified as a dependency of `st3375` so it should be installed in all cases.

Generally `pip install pillow` should *not* be triggering a build from source and suggests a configuration error with your Pi - `/etc/pip.conf` should include piwheels like so:

```
[global]
extra-index-url=https://www.piwheels.org/simple
```

If you're having to use apt, something is fundamentally broken!

closed 2025-03-18 by Gadgetoid

slokhorst, 2025-03-23:
Fresh Raspberry Pi OS Lite (32-bit) (Bookworm) install on a Pi Zero W:

```bash
sudo apt update
sudo apt upgrade
sudo apt install git

git clone https://github.com/pimoroni/enviroplus-python
cd enviroplus-python
./install.sh
sudo reboot

source ~/.virtualenvs/pimoroni/bin/activate
python ~/Pimoroni/enviroplus/examples/all-in-one.py
```

```
Traceback (most recent call last):
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/_core/overrides.py", line 7, in <module>
    from numpy._core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/__init__.py", line 114, in <module>
    from numpy.__config__ import show_config
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/_core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/sebas/.virtualenvs/pimoroni/bin/python"
  * The NumPy version is: "2.2.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/sebas/Pimoroni/enviroplus/examples/all-in-one.py", line 7, in <module>
    import st7735
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/st7735/__init__.py", line 26, in <module>
    import numpy as np
  File "/home/sebas/.virtualenvs/pimoroni/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
```