pimoroni/enviroplus-python #146: LCD doesn't work with last drivers/setup

idotj, 2024-11-25:


Gadgetoid, 2024-11-25:
You should probably update to Bookworm. Bullseye is effectively broken for GPIO and, iirc, the pin labels are probably PINXX rather than GPIOXX.

If you `sudo apt install gpiod` and run `gpioinfo` it should show you the available `gpiochipN` devices and their "lines" (which usually correlate to physical pins)

You can then, potentially, just swap "GPIO9" out for the relevant named pin on your device and it should still work.

idotj, 2024-11-25:
Thanks for the quick reply!  
Looks weird but it works if I use only the pin number as string (tried with PIN and GPIO prefix and it didn't work).

In my case for **lcd.py** it will work if I set:

dc="09"
backlight="12"

But if I use that config, in **all-in-one.py** it will show again the same error:

Woah there, suitable gpiochip not found!
  ❌  O9: not found - /dev/gpiochip0 (pinctrl-bcm2835)!

So that means that for each project the pin number will jump??
Sorry but for hardware/network reasons I can't install Bookworm.

Gadgetoid, 2024-11-25:
`O9` or `09`?

idotj, 2024-11-26:
Sorry for the typo! (good catch)
It works as expected. I hope the search engines will index this issue, just in case someone else has the same problem can fix it using:

_dc="09",
backlight="12",_

Instead of:

_dc="GPIO9",
backlight="GPIO12",_

idotj, 2024-11-26:
BTW, thanks for your quick reply and reactivity.
All the best!

Gadgetoid, 2024-11-26:
> good catch

 I started out life in digital and print media 😆 
 
 And thank you, I'm glad you're up and running!