pimoroni/enviroplus-python #148: LCD 0.96" is not working in latest stable release

gilbertfrancois, 2024-12-31:
## Description

In the current stable release `v1.0.2` and `main`, the LCD display is not working. When I run e.g. the example `lcd.py`, I only see the backlight switching on, but no image appears. Same for all other examples, like `all-in-one.py` and more. I noticed that before running lcd.py script, the `pinctrl` shows:

```
 9: a0    -- | lo // GPIO9 = SPI0_MISO
10: a0    -- | lo // GPIO10 = SPI0_MOSI
11: a0    -- | lo // GPIO11 = SPI0_SCLK
12: op -- -- | lo // GPIO12 = output
```

After running the script, the `pinctrl` command shows:

```
 9: op -- -- | hi // GPIO9 = output
10: a0    -- | lo // GPIO10 = SPI0_MOSI
11: a0    -- | lo // GPIO11 = SPI0_SCLK
12: op -- -- | lo // GPIO12 = output

```

GPIO 9 is no longer assigned as SPI MISO. 

I tried changing the initialization of the ST7735 class like suggested https://github.com/pimoroni/st7735-python/issues/12#issue-811277309, but that didn't resolve the problem:

```python
# Create LCD class instance.
disp = st7735.ST7735(
    port=0,
    cs=st7735.BG_SPI_CS_FRONT,
    dc="GPIO9",
    backlight="GPIO12",
    rotation=270,
    spi_speed_hz=10000000
)
```

Another suggestion from https://github.com/pimoroni/enviroplus-python/issues/146:

```python
# Create LCD class instance.
disp = st7735.ST7735(
    port=0,
    cs=1,
    dc="09",
    backlight="12",
    rotation=270,
    spi_speed_hz=10000000
)
```

But also no success...

## Steps to reproduce

- Create new image with Rasberry Pi imager, using Raspberry Pi OS lite, 32bit, bookworm
- Once logged in and before checking out enviroplus-python, run commands below to make the installer happy:

```sh
# Make static link for /boot/config.txt.
sudo mv /boot/config.txt /boot/config.bu.txt
sudo ln -s /boot/firmware/config.txt /boot/config.txt

# Install PIL and numpy, since they are missing in the dependencies of install.sh and depend on non pure python libraries.
apt update
apt install git python-pil python-numpy
```

- Follow instructions from this page: [[Installing the Enviro+ Python library](https://learn.pimoroni.com/article/getting-started-with-enviro-plus#installing-the-enviro-python-library)](https://learn.pimoroni.com/article/getting-started-with-enviro-plus#installing-the-enviro-python-library	)

[... 223 lines cut ...]
dtoverlay=pi3-miniuart-bt
Adding dtoverlay=adau7002-simple to /boot/firmware/config.txt
dtoverlay=adau7002-simple

Would you like to copy examples to /home/pi/Pimoroni/enviroplus? [y/N] y
Copying examples to /home/pi/Pimoroni/enviroplus
Done!

Would you like to generate documentation? [y/N] n

All done!
If this is your first time installing you should reboot for hardware changes to take effect.
Find uninstall steps in /home/pi/Pimoroni/enviroplus/uninstall.sh

```

## Possible similar issues

https://github.com/pimoroni/st7735-python/issues/12

https://github.com/pimoroni/enviroplus-python/issues/146

https://github.com/pimoroni/enviroplus-python/issues/128

https://github.com/pimoroni/enviroplus-python/issues/97

nicblockley, 2025-01-05:
I am experiencing the same problem too. The examples which output the sensor measurements to the CLI are working fine. I followed another issue post to install numpy and pil via apt install which allows the examples that rely on the LCD to run, but the LCD doesn't display anything.

VictorJJim, 2025-03-09:
Hi, is there any solution to this issue? I'm seeing it as well.

Erik4556, 2025-03-14:
Also experiencing this issue, Ras Pi Zero 2 W, 64bit Bookworm, nothing but PMS5003 installed.

Gadgetoid, 2025-03-18:
I've tested on Pi 4 and Pi 5 with fresh installs and the LCD works in both cases.

Please try:

```
pip install --ugprade enviroplus
pip install --upgrade st7735
```

Please also provide contents of `/boot/firmware/config.txt` lest some other setting is conflicting.

Erik4556, 2025-03-18:
@Gadgetoid 

I have run the above and can confirm on my pi zero 2 W there is no change in behavior. Backlight illuminated, no display. I have attached my config.txt here

[config.txt](https://github.com/user-attachments/files/19324045/config.txt)

Original poster was using a Pi Zero W, potentially something related to the pi zero itself that's not being considered here?

Gadgetoid, 2025-03-18:
> potentially something related to the pi zero itself that's not being considered here

Very possible, I assumed this but tried and failed to get a Pi Zero W to work. I'll have to avail myself of a new one and try again.

Gadgetoid, 2025-03-21:
Well it wasn't here that I mentioned it, but I shall have to eat my words regarding libopenblas.so.0 which *is* required on Pi Zero W for reasons I cannot fathom. It also requires `libopenjp2-7`. (Notably Pi Zero W has its own OS release which is not cross compatible with other versions.)

For posterity more than an answer to this issue the fix is: `sudo apt install libopenblas0 libopenjp2-7` which gets a virtualenv installed version of pillow/PIL working.

After setting up my Pi Zero using Raspberry Pi OS Lite (*not* Legacy) both the display and PM sensor are working.

Other than running `install.sh` as it currently stands, and answering y to installing example dependencies I only needed the above extra libs...

I'm going to run an upgrade/update since I can see a new kernel and such in there, maybe that breaks it 🤷

As for `GPIO 9 is no longer assigned as SPI MISO.` that's just the MISO pin (device -> host) being repurposed as a data/command signal, which is common for displays because they are often write-only. It's as expected.