pimoroni/enviroplus-python #150: One line installer works, but errors abound

EpicallyJess, 2025-01-14:
I get errors for practically all the sensors when trying to run the examples in the virtual enbironment

bme280, st7735, pms5003 all throw exceptions and say that packages aren't installed
got an error for the roboto font, something called PIL, and more.

I don't know python

Fresh bookworm (lite) and bullseye (lite) installs.  literally created a user, installed git, cloned, installed, then activated virtual environment.  Can't run a single example.

thirdr, 2025-01-15:
Hello :) 

First thing to check is that you're in the correct virtual environment. You should see `(pimoroni)` to the left of your username. If you don't see this then you can run the command `source .virtualenvs/pimoroni/bin/activate` from your home directory to get into it.

Once you've got that, you can install PIL inside your virtual environment by running the command `pip install pillow`. This will install all the necessary files.

You should then have everything you need to run the examples provided. It's worth mentioning that you'll need to use the command to activate the virtual environment each time you reboot if you want to use the examples for enviro+

EpicallyJess, 2025-01-15:
I've ensured that I'm running in the virtual environment and installed Pillow.  Other errors still persist.  I'll try and get screenshots but might take some time.

jacotronix, 2025-01-15:
I found that I had to ensure the user was in the dialout, i2c, gpio & spi groups. After running the install script, the user was not in them all.

Sorry I can't remember which as I'm developing my own environment which will simply support my uses rather than having to cater for everyone.

Gadgetoid, 2025-03-18:
I've merged a fix for the missing "PIL" (the pillow library), you should answer 'y' to installing example requirements or run:

```
pip install -r requirements-examples.txt
```

A couple of other fixes have been merged too, and I've had no other issues with fresh installs on Pi 4 or Pi 5.

Feel free to re-open if you encounter problems (and please provide details/screenshots so I can tackle them)

Note: bookworm is required, I don't test on anything prior since Pi 5 needs it and the new libgpiod method of GPIO access means supporting prior versions is a nightmare.

closed 2025-03-18 by Gadgetoid